KiCad/kicad-doc #372: Pcbnew: Sections 10.9 and 10.10

evanshultz, 2016-06-12:
A few issues here:

1. There appears to be a toolbar icon missing in the first sentence of 10.9.1 ("...using the button ."). It should be the "Add filled zones" icon.

2. Both dialog screenshots use French layer names, which isn't an issue per se but I assume it should be English like other screenshots I see.

3. The screenshot in 10.9.1 is an old dialog and not representative of 4.0.2 stable.

ciampix, 2016-06-12:
Thanks for the heads-up. If you are so kind to deliver the missing / right icon and screenshot or please be patient and I will do it...

evanshultz, 2016-06-12:
I don't have write access, but the new images are attached and need to go in kicad-doc/src/pcbnew/images/.
[image]
[image]

commit c74729d3 (2016-06-13): Updated screenshots to Pcbnew Manual, thanks to evanshultz, closes #372
closed 2016-06-13 by ciampix