KiCad/kicad-doc #378: Some images (screenshots) are installed on multiple lang/directories

jsreynaud, 2016-06-15:
I took one example:
File dialog_bom.png is in source src/eeschema/images/<lang>/

Example with md5:
a4a8ce69e008c1df7a075ec4d4ec6164 src/eeschema/images/fr/dialog_bom.png
b42a61bdc4c40b2f16bc2dccf90759d4 src/eeschema/images/en/dialog_bom.png
c10f66ad10b52eae1d803a2254d7608e src/eeschema/images/ja/dialog_bom.png

There is one file for one language (this is correct)

But when it's installed onto /usr/share/doc/kicad/help, "en/dialog_bom/png" is installed for all other languages too:
cd /usr/share/doc/kicad/help
md5sum */images/en/dialog_bom.png
b42a61bdc4c40b2f16bc2dccf90759d4 de/images/en/dialog_bom.png
b42a61bdc4c40b2f16bc2dccf90759d4 en/images/en/dialog_bom.png
b42a61bdc4c40b2f16bc2dccf90759d4 es/images/en/dialog_bom.png
b42a61bdc4c40b2f16bc2dccf90759d4 fr/images/en/dialog_bom.png
b42a61bdc4c40b2f16bc2dccf90759d4 it/images/en/dialog_bom.png
b42a61bdc4c40b2f16bc2dccf90759d4 ja/images/en/dialog_bom.png
b42a61bdc4c40b2f16bc2dccf90759d4 pl/images/en/dialog_bom.png

It look to be linked with issue #132
Thanks,

BrianSidebotham, 2016-08-31:
This is a duplicate of #170 as far as I can see...

nickoe, 2016-09-28:
Closing as duplicate for #170

closed 2016-09-28 by nickoe