KiCad/kicad-doc #393: IDF Exporter separate from Pcbnew docs?

evanshultz, 2016-06-21:
Why is https://github.com/KiCad/kicad-doc/blob/master/src/idf_exporter/idf_exporter.adoc not part of the Pcbnew documentation? I don't see it linked from anywhere in the KiCad toolset, so it seems users aren't normally exposed to this document? Should it be merged with Pcbnew's documentation?

There are a number of bugs explicitly mentioned in the IDF Exporter documentation, and I assume the status of those bugs is unknown?

Is it acceptable for me to review bugs, update the doc separately, and then propose a merge into Pcbnew once it's ready?

ciampix, 2016-06-21:
Good point. Fresh minds quickly see the holes.
As usual, this is due to historical reasons. It is so because the IDF_Exporter was developed as an external project and then merged into KiCad. Probably it should be merged with the Pcbnew documentation or, at least, referenced from this.

nickoe, 2016-06-21:
Yeah, well, this was mainly because that it needed to be documented, but it was not entirely clear where to add it at the time, so it was just added as a seperate document as a start.

evanshultz, 2016-06-22:
Excellent. Will do.

closed 2016-06-22 by evanshultz