KiCad/kicad-doc #394: GSIK: Proposal for restructure

evanshultz, 2016-06-21:
Currently there are a few sections at the top which are nice, IMO. But after that, it gets unclear to me:
-There's a huge Eeschema section which covers project creation, exporting a netlist, assigning footprints, generating a BOM, etc. It's monolithic.
-Next is the Pcbnew section, which just covers designing the PCB. Generating and checking Gebers is a separate section, as is auto routing and annotation.
-Then is the creation of schematic parts by a number of means and then making footprints. And finally a small note (which is a somewhat of a repeat of the Support section at the top).
-Finally there are some notes.

The flowchart isn't especially clear, because it doesn't show project creation, shows two different ways to assign footprints, etc. (There are instructions to update this, I know.)

It seems to me that a few more sections would be helpful in showing the process, mostly on the schematic side. Here's my suggestion for re-arranging this doc starting at the "Draw electronic schematics" section (section titles not finalized, just the workflow):
-Create project (KiCad manager)
-Eeschema
--Draw schematic
--Bus connections
--Annotate schematic
--ERC
--Assign footprints
--Generate netlist
--BOM
-Pcbnew
--Set up board (DRC, layers, etc.)
--Import netlist
--Place components
--Lay out copper
-FreeRouter (see below)
-Design changes (Backward Annotation also included here)
-3D viewer
-Generate fab files (back to Pcbnew)
-View fab files
-Making components (few sections)
-Making footprints
-Send off

What to do with FreeRouter? Considering that project's current status, does this section get removed? Keep it as is, with polish as needed? Add note about Java version/dependencies?

I also find that titles using a present participle (-ing words) are more friendly, so I suggest changing "Generate Geber files" to "Generating Geber files". And also eliminating first person ("we, "let's") in favor of strictly second person ("you", "your") perspective.

Is there something wrong with this? It seems to me that breaking things down a bit more is better to show the process. Comments? Criticism?

kinichiro, 2016-06-22:
Here is my personal thoughts.
I always would like to balance the jobs between writers and translators.
Good modification brings better docs, on the other hand, localized docs go back to English again.
Recently, before this issue was posted, many kinds of translators updated their language versions of GSiK and other manuals and posted many PRs here.
Reconstruction of doc has big influence on translation achievements.
I don't mean discouraging writers. I just want to balance both.
I will think about this issue more.

ciampix, 2016-06-22:
I agree. Updating translation on an always moving target of reference doc can be frustrating. I know of at least a translator that abandoned the KiCad translation team because the changes on the reference English docs invalidated much of his translation work. We do not want this to repeat again so it is better that we balance between doc writers search for perfection and translators work. One way to do it is to revise docs not too often. Probably we need to establish a sort of "freeze" when we think English reference docs are satisfactory. We have git and his branch function. Let's use it. One thing I think IMHO is wrong is that "rolling docs" are the reference on-line. We should have instead "stable" docs for the stable KiCad version, and translators should contribute to the "stable" branch. Rolling translations _could_ (and I think _should_) be on line too, but it must be clear that they refers to the devel version and they are subject to change (translators beware!). What do you think?

evanshultz, 2016-06-22:
Ooohhh. Thank you. All good points. I hadn't thought about translation.

I should have been more clear that my core idea was the addition of sections to break up the existing content. I wasn't thinking about creating new content. 56 steps is a lot, especially when only about half of it truly falls in the section title of "draw electronic schematic".

I'll add in the new sections to see what it looks like, then create a pull request for everyone's review. IMO it will help to better define the process and make GSIK more palatable for new users.

I like the idea of online documentation that is easily updatable, especially as Wayne has already shown he will release more frequent updates to KiCad. There are a number of challenges here, and I don't have the experience to know what would work best for the KiCad team and, most importantly, the users. Thanks for teaching me and being so accepting of my ideas and contributions.

ciampix, 2016-06-23:
I want to be clear here: your work is tremendous and I thank you very much for it. No question about it. I was only expressing a problem that emerged that perhaps has not a real catch-all solution.

kinichiro, 2016-06-29:
>We should have instead "stable" docs for the stable KiCad version, and translators should contribute to the "stable" branch.

@ciampix I like your idea with using git branch function.
- On `stable-4.0` branch, we don't modify .adoc, and translators contribute here, and release pkg docs from this branch.
- On `dev-5.0` branch, doc writers update .adoc
- When the new software release period (i.e. KiCad 5.0) come, create new branch `stable-5.0` from `dev-5.0`, copy .po files from `stable-4.0` branch, and translators contribute here.

I would like to hear from @nickoe, too.

evanshultz, 2016-07-07:
Please see the patch I just submitted. It should be taken as a guide which is knowingly unfinished. Once we're happy with the flow of the sections, I'll fix the numbering and edit the text to smooth together the sections before submitting a final patch that is suitable for merging.

Remaining questions:
1. Bring everything under Tutorial "down" a level, with Portability and Support going back "up"? It will mean we're using all 4 levels and give less depth to each title, but I prefer it.
2. FreeRouter before or after Forward Annotation? I say after since usually during placement you'll see something to change, with an autorouter being the last step after placement is finished.
3. What to do with FreeRouter? Considering that project's current status, does this section get removed Keep it and add note about Java version requirement?

kinichiro, 2016-07-07:
I think there are 2 ways to split into writer's branch and translator's branch.

(1) `master` branch for a writer, branch-by-version for a translator.

For example, branch `v4.0.0` is created, and translators submit PR to this branch,
writers submit PR to `master` branch.

Next new translation season come,
At first, we just create new branch `v5.0.0` from `master`,
and copy .po files from `v4.0.0` branch to `v5.0.0` branch,
then tell all translators "Please translate on branch v5.0.0".

This method seems natural, but some negative points;
- (OK) writer just submit PR to `master`
- (NG) translators need to submit `specific translation targeted branch`
- (NG) KiCad web site should build docs from this `specific translation targeted branch`, and it will change next release season.
- (NG) binary packager should do the same thing as KiCad web site.

(2) `master` branch for a translator, branch-by-version for a writer.

For example, branch `v5.0.0` is created, and writers submit PR to this branch,
translators submit PR to `master` branch.

Next new translation season come,
At first, we create branch 'v4.0.0' from 'master'.
Then cherry-pick all writers' commits from `v5.0.0` to `master`
and tell all translators "Please translate on branch master".
Finally, we create new branch `v6.0.0` from `master` for writers.

This method seems tricky a little bit, but has some good points;
- (NG) writer need to submit `specific writer branch`
- (OK) translator just submit PR to `master`
- (OK) KiCad web site can build docs always from `master`, this is same as current operation.
- (OK) binary packager can do the same thing as KiCad web site.

I would like to suggest method (2), since `master` branch always has stable translated docs,
and KiCad web site and binary packagers can focus on 'master' branch only.

BTW, if we create branch `v4.0.0` at Jun 15, 2016 commit https://github.com/KiCad/kicad-doc/commit/ab01c5f5d3d9ea518802966285c7f41107de8d08 ,
we are able to resque almost all the translation works as 4.0 docs.

ciampix, 2016-07-07:
I _really_ like method (2). Great idea. I knew that there should be a way to resolve the translators issues just using git better, and you @kinichiro focused on it precisely!

ciampix, 2016-07-07:
I would like to make arrangements to create automatic builds also for the dev branch, just to be able to check the results in real-time for errors...

ciampix, 2016-07-07:
Having packaged and on-line just the master branch with the on-going translation is ok for me...

kinichiro, 2016-07-07:
@evanshultz 
sorry for off topic discussions here a lot, I created another issue for that and moved.
And do you mean this ? https://github.com/KiCad/kicad-doc/pull/407
Please take consider it takes a time to review since your changes are huge.

evanshultz, 2016-07-07:
No problem. This is the way version control is supposed to work. I hope to help get everything sorted out with you guys.

I'm looking forward to your review/feedback when time allows, and I'll find other things to do in the meantime. :)

evanshultz, 2016-07-11:
Please note that PR #407 should only need section flow and section name review now. Do not review content, as there are obvious breaks where sections were added or moved. That should limit the amount of review needed for GSIK before we can agree on the structure; after that I will move into patches to fit the content into the new structure before the PR can be finished and merged.

kinichiro, 2016-07-14:
I built your PR to PDF, and see the contents order and structure.
Here is your suggested structure of GSiK, right ?

```
1 Introduction to KiCad

2 Downloading and installing KiCad
  2.1 Under GNU/Linux
  2.2 Under Apple OS X
  2.3 Under Windows

3 Using KiCad
  3.1 Shortcut keys
    3.1.1 Accelerator keys
    3.1.2 Hotkeys
    3.1.3 Example

4 KiCad Workflow
  4.1 Overview
  4.2 Forward and backward annotation

5 Tutorial

6 Creating a new project

7 Using Eeschema
  7.1 Drawing a schematic
  7.2 Connecting to a bus
  7.3 Annotating the schematic
  7.4 Electrical rules check

8 Assigning footprints

9 Saving a netlist

10 Generating a BOM

11 Using Pcbnew
  11.1 Creating and setting up a PCB
  11.2 Designing a PCB

12 Forward annotation in KiCad

13 Autorouting with FreeRouter

14 Generating Gerber and drill files

15 Using GerbView

16 Making schematic components
  16.1 Using Component Library Editor
  16.2 Exporting, importing and modifying library components
  16.3 Making components with Quicklib
  16.4 Making a high pin count component

17 Making component footprints
  17.1 Using Footprint Editor

18 Portability of KiCad project files

19 KiCad support and documentation
```

IMHO, (so you don't need modify your PR according to my opinion, let's hear from others, too.)

- too many chapters for tutorial, and I feel the story is broken in chunk.
  e.g. jobs done by Eeschema are in chapter 7, 8, 9 and 10

- chapters for continuous jobs are far away
  e.g. "11. Using Pcbnew" and "14. Generating Gerber..." and "15. Using Gerbview"
  I think these are the continuous jobs, personal feelings, though.

> 1 Bring everything under Tutorial "down" a level, ...

IMHO, GSiK should have Informational chapter, Tutorial chapter and Tips chapter, then tutorial sections go under Tutorial.
And tutorial chapter should be shorten, sections below could move to Tips section.

```
  7.2 Connecting to a bus
  10 Generating a BOM
  12 Forward annotation in KiCad
  13 Autorouting with FreeRouter
  16 Making schematic components
  17 Making component footprints
```

> 2 FreeRouter before or after Forward Annotation? ...

IMHO, It is OK if these 2 sections are splitted from main tutorial story.

> 3 What to do with FreeRouter?...

IMHO, I would like to keep it. FreeRouter project status seems not active, but KiCad users often use it.

I would like to hear from others, too.
Thanks