KiCad/kicad-doc #400: OS X support is experimental?

evanshultz, 2016-06-27:
https://github.com/KiCad/kicad-doc/blob/master/src/kicad/kicad.adoc mentions this, but the three OS options seem to be mentioned with equal footing on the website and elsewhere I've seen. Is this true, or legacy text that can be removed?

nickoe, 2016-06-27:
@evanshultz Thank you for spending effort in keeping the docs recent!

Yes, I think we can proudly say that OS X is supported by now. There are multiple developers on OS X and it seems that it is on par with the other platforms since kicad 4.0.0.

ciampix, 2016-06-27:
Thank you @nickoe, I was just waiting a dev word on this. Ok @evanshultz please go on and thank you for this!

evanshultz, 2016-06-28:
And there we have it!

closed 2016-06-28 by evanshultz