KiCad/kicad-doc #403: Pl_Editor restructure

evanshultz, 2016-06-28:
https://github.com/KiCad/kicad-doc/blob/master/src/pl_editor/pl_editor.adoc jumps right into doing stuff after the intro. It feels abrupt. Later on we find the main window description, toolbar introduction, etc.

I propose this structure (with abbreviations for existing section names):
-Introduction (include Files and Invoking sections, remove section about element type as it's below)
--Basic properties
--References and Coordinates
-Interface (was Commands)
--Main window (include Status bar)
--Main toolbar
--Left window
--Right window
---Keyboard
---Mouse
--Selection
-Adding Elements
--Creation (merge Context Menus)
--Lines and rectanges
--Text (include Page 1 option)
---Symbols
---Multi-line (including in page setup)
---Constraints
--Bitmaps
--Logos
--Rotation
--Repeat

The document flow will then follow:
-talk about what Pl Editor is and does
-brief description of page layout concepts
-orientation to the Pl Editor interface
-general interaction principles for all elements
-adding and manipulating specific elements

To be clear, I'm talking about moving the existing sections around and not changing the content of the documentation. "Merging" or "including" above just means smash the content of those two sections together, with cleanup and reformatting to be a later step based on our discussion of GSIK where translation issues are a legitimate concern.

Agree? Disagree? Comments?

kinichiro, 2016-06-29:
I agree.
The structure you mentioned above is as same as other reference manuals,
such as eeschema ,pcbnew, kicad manager and gerbview.
I believe the reference manuals should follow the same structure.
And I also appreciate your consideration for translators works.

Best regards,

evanshultz, 2016-07-11:
PR #409 submitted for this issue. I assume discussion will continue in the PR comments, so this is just a comment that progress is being made.