KiCad/kicad-doc #408: separating branches for writers and translators

kinichiro, 2016-07-07:
As discussions at https://github.com/KiCad/kicad-doc/issues/394 , I would like to design the branch separation, here,
based on method (2)**`master` branch for a translator, branch-by-version for a writer**
described at https://github.com/KiCad/kicad-doc/issues/394#issuecomment-231078651

ToDo
- create simple rule and procedure for this.
- resque the past translation works as 4.0 docs. (e.g. by Jun 15, 2016 commit)
- make arrangements to create automatic builds also for the dev branch, just to be able to check the results in real-time for errors
- Add announcement to README.adoc, "Contributing" section, "Writer should contribute to branch v5.0.0, Translators should contribute to master branch."

Issue
- for the first time, we need to remove commits newer than e.g. Jun 15, 2016 from `master` after creating branch `v5.0.0` for writers. This affects everyone clones this Github repository.

kinichiro, 2016-07-08:
**Draft for procedures and rules**
- (I will update this description in this comment.)
- (I suppose to put this description into README.adoc or something.)

---------+---------+---------+---------+---------+---------+---------+---------+
== Separating branches for writers and translators

Documentation work consists from writers job and translators job.
The writers updates English adoc files and English screen shot png files.
The translators updates localized po files and localized screen shot png files.
First of all, the writer creates adoc in English, then the translator generates po and translate it.
After its translation is completed, if the writer updates the adoc, that part of translations will be invalidated.

To protect the translated docs while updating original English docs, our team seprates branches for writers and translators.
By this operation, translators are able to do their jobs without worrying about invalidation by changing referenced English docs, and writers can update original English docs according to software update without breaking translated docs.

=== Branching Procedures 

Here is overview of this procedures.

```
                 Next release time|
                                  V
                                 <save_(LAST)>
<save>                           [*]
                                  ^
                         1.branch |
<master>  [+]--------------------[+]  -  -  -  - {translators}{web}{pkg}
             \                    ^ \
              V           2.merge |  V 3.branch
<dev>         [#]----------------[#] [#]  -  -  -{writers}
              <dev_(CURRENT)>        <dev_(NEXT)>
```

1. branch `<save_(LAST)>` from `<master>` for saving past work
2. merge `<dev_(CURRENT)>` into `<master>` to obtain all adoc updates, and translators begin their work here
3. branch `<dev_(NEXT)>` from `<master>` for writers

=== Rules

By Contributors

- *{writers}* should work on branch `<dev>_(NEXT)`, and submit PR to here.
- *{translators}* should work on branch `<master>`, and submit PR to here.
- *{web}* and *{pkg}* obtain docs from branch `<master>` for the latest translated docs.

By Branches

- On branch `<master>`, *.po* and localized screen shot *.png* should be updated.
- On branch `<dev>`, *.adoc* and English screen shot *.png* should be updated.
- On branch `<save>`, nothing should be changed since this is for the historical record.

kinichiro, 2016-07-08:
**Draft for transition steps and rescue the translation works as 4.0 docs. (e.g. by Jun 15, 2016 commit)**
- (I will update this description in this comment.)
- (This transition steps will be executed 1 time only.)

---------+---------+---------+---------+---------+---------+---------+---------+
Transition Steps.

Currently, our branch is like this,

```
                                |we're here, now
                  Jun 15th      V
<master>   ----------+---------[+]  -  -  -  -  -{translators}{writers}{web}{pkg}
```

At first, we should change branches as follows,

```
                                |we're here, now
                  Jun 15th      V
<master>   ---------[+]  (2.remove commits)  -  -{translators}{web}{pkg}
                                 \
                         1.branch V
<dev>                             [#]  -  -  -  -{writers}
                                  <dev_5.0.0>
```

Then, v5.0.0 release time has come, our procedures should be like below,

```
               v5.0.0 release time|
                                  V
                                 <save_4.0.0>
<save>                           [*]
                                  ^
                         1.branch |
<master>  [+]--------------------[+]  -  -  -  - {translators}{web}{pkg}
             \                    ^ \
              V           2.merge |  V 3.branch
<dev>         [#]----------------[#] [#]  -  -  -{writers}
              <dev_5.0.0>            <dev_6.0.0>
```

evanshultz, 2016-07-08:
A few thoughts (which ended up growing out of control as I typed [sorry!]):

1. It feels to me like this is a new concept, but it's fundamental version control. It's why version control exists. See https://git-scm.com/book/en/v2/Git-Branching-Basic-Branching-and-Merging. In Issue #394 I thought you were simply talking about *using* Git features, not wondering if it can be done. IMO, there's no need to show these basic processes in a "flowchart"; just describe what to do.

2. Only one `dev` branch may not be enough. Once a branch is merged to `master` at the end of a cycle, it goes away. The flowchart shows `dev` is then re-created as a new branch that is ready for contributions. I guess you used `dev(CURRENT)` and `dev(NEXT)` just for clarity, but we should support multiple branches in parallel instead of one in series. What if we write documentation for 4.2 right now, but there are 4.1 features that also need to be added? They can't both go into a single `dev` branch because we need to build docs for 4.1 and not 4.2. Or consider if a feature is intended for a release but it slips and is pushed out to a later release. We need somewhere for that doc/patch to go while we wait for the feature to ship. We will need to be clear in README about this process so that contributors have a chance of knowing what the branches are for and where to submit their PRs, and also clear about our method for creating branch names. (It's possible we decide to just use one branch and let docs/PRs for future not get merged into `dev`, so that when `dev` is merged to `master` we can then include the docs/PRs in a future `dev` branch, but that goes against the principles of version control. And it also doesn't utilize version control's power. So we need to plan for this now, IMO.)

3. Be careful about tying any step to a KiCad release. For major versions we will surely have new documentation. Minor releases may or may not. Patch releases likely will not. The web docs can be updated at any time. What if we decide to stop including doc files with the release? Wayne has shown a willingness to make KiCad more evergreen, and so I think we need to be flexible. I'm not saying anything above is wrong or needs to change, but we must consider out-of-band as well as synchronized doc releases.

4. Choosing where to merge is based on purpose. For example, right now corrections/typos could be made for 4.0.2 which affects `master` and all `dev` branches, or it could be new content for 4.1 which goes into a dev branch but is merged before commits related to 5.0. So the `dev` branch isn't the only one that may receive commits, either before or after a release. Even a typo that affects `master` and any dev branches could trigger new doc generation since at least the web docs could be corrected.

5. I use SVN regularly and Git only rarely, but the concepts are similar and comfortable. I'll try to figure out the Git commands for the transition this weekend and post them here.

6. And on that note, since 15 Jun the changes are still relevant for 4.0.2 and will be largely, if not completely, for the next release. When do we make this transition? Should we do it at a patch or minor release, like 4.0.3 or 4.1 (if there will be one), so that we get a chance to try it out and address issues when things are relatively calm? I suspect 5.0 will be a busier time and may be well into the future.

7. After agreeing on a plan, I think we should address the writing style guide. It will give us a better foundation for future documentation work, and also close that PR which none of us own. I can re-create my outstanding PRs to the new dev branch.

8. @kinichiro I like the draft you made for README, except I question the flowchart as mentioned above. I can't edit your post, but I see a couple small things I'd like to bring up. How best to do that?

kinichiro, 2016-07-08:
1. I would like to share the concept with someone who doesn't know well about Git and Github, too.
   Just description is not enough, and we worked and are going to work together with any skill level contributor.

2. I don't mention that `dev` branches are deleted after merging. We can keep them.
   But, I don't think we can manage such many versions.
   Now I'm thinking about 1 translation branch (`master`) and 1 dev branch (`dev`), only.
   Because we don't have human resources to maintain multi versions docs.
   For `dev` branches, YES, you can have multiple branches `dev_4.0.1`, `dev_4.0.2`, `dev_4.0.3`, ...
   But, we can't translate them for each versions !

3. Currently, KiCad software has `4.0` Bazaar branch and `product` Bazaar branch.
   I suppose `4.0` is tying to `master` of doc, `4.0` English docs are finished to write, only translation works are remain.
   And `product` is developement branch for the next release (5.0.0), it is tying to `dev` of doc, writers will follow new feature and describe it to `dev` branch.
   `4.0` might provide new feature developed at `product` that is not described in docs by minor release, though, I gave up that.
   That kind of new features should be written in `dev` branch.

4. IMHO, HEAD of latest `dev` branch will be merged to HEAD of `master` branch.
   And merging will be done at next major release of KiCad.
   Merging `dev` to `master` means that we ask translations for the New Release of KiCad.

5. Thanks, but I wrote git commands in the drawings above, though, 
   `(2.remove commits)` will be done by git rebase or something.

6. I believe any time is OK, and patch or redistribution are not needed.
   Have you ever seen other language docs on http://kicad-pcb.org/help/documentation/ ?
   There are current versions of docs there, and some of them were translated but now some parts are invalidated.
   I just want to rescue them.

7. I would like to progress this separation after resolving your PRs.
   I don't want you to re-submit your PRs for repository management reason.
   And I believe this takes a long time to adjust.

8. I can't tell what is your couple of small things, though, any way is appreciate, briefly pls.

nickoe, 2016-07-11:
I have been away the last week, but have you had a look at https://github.com/KiCad/kicad-doc/blob/master/docs-versioning.adoc ?

nickoe, 2016-07-11:
@kinichiro  What commits do you mean are needed to be removed and why?

kinichiro, 2016-07-12:
@nickoe I had read versioning text.
And I think that policy doesn't work correctly to protect from invalidation
of translated docs, then I post this.

> What commits do you mean are needed to be removed and why

I meant that commits after e.g. Jun 15th should be removed from the master
branch only at this time, since they are invalidate the translations.
But, those commits are still in dev branch, and will merged in future.

I would like to rescue translated 4.0 docs, "rescue" means published as
formal docs, here.
Formal published docs should be shown on the KiCad web site and binary
distribution includes.
At this time, KiCad web site introduces docs but they are the latest
developing ones not the stable translated ones.

nickoe, 2016-07-12:
Updates for kicad 4.0 should be made on the 4.0 branch, see https://github.com/KiCad/kicad-doc/tree/4.0

I don't think we should have a branch named dev as you suggest. This is what the master branch is for, the master branch in kicad-doc and kicad-i18n is supposed to track the product branch of lp:kicad.

Translators are encouraged to translate on the stable branches, for example 4.0. So the task right now is to make sure that the 4.0 tip is up to date such that we can very soon tag a 4.0.3, which can be released together with the 4.0.3 release of the kicad. I don't think we should backport the restructuring we have done on master to 4.0, we should just consider "bug fixes" as we do on the product branch of lp:kicad.

If translators are bored or whatever I won't keep them for translating on the master branches, but they should of course keep in mind that the translations have a high risk of becoming outdated.

kinichiro, 2016-07-12:
Then,
will KiCad web site publish docs from 4.0 branch of kicad-doc ?
Does 4.0.x stable binary package includes docs from 4.0 branch ?

If so, we should maintain 4.0 branch as the stable docs branch.
I think that is able to do with cherry-pick ing from master branch.

And from now, we should announce to translators to submit PRs to 4.0 branch
not master branch.
Also, encourage writers to submit PRs to master branch.

1 question,
What do you use tag for ?
I think 4.0.0 doc and 4.0.3 doc are the same, since minor releases are just
bug fixes and no new features are added to the software. Then docs are the
same too. No?

kinichiro, 2016-07-12:
Current status of kicad-doc stable `4.0` branch here,
and tag `4.0.1`, `4.0.2` and branch `4.0` is the same.

branch
4.0 latest commit : 2015-12-10 https://github.com/KiCad/kicad-doc/commit/ffa1524a9007a1bc5544b4b3037eeb5dfd54e1a8

tag
4.0.0-rc1 : 2015-09-12 https://github.com/KiCad/kicad-doc/commit/4d7142b2aeb735b48af49c0d90f52d795c604055
4.0.0-rc2 : 2015-11-08 https://github.com/KiCad/kicad-doc/commit/ccf13bf7a9954c84544a9536e82bf7e38e0c489d
4.0.0     : 2015-11-26 https://github.com/KiCad/kicad-doc/commit/b7726747065b11a0e6b3edcb8695decee216ec9e
4.0.1     : 2015-12-10 https://github.com/KiCad/kicad-doc/commit/ffa1524a9007a1bc5544b4b3037eeb5dfd54e1a8
4.0.2     : 2015-12-10 https://github.com/KiCad/kicad-doc/commit/ffa1524a9007a1bc5544b4b3037eeb5dfd54e1a8

I would suggest that we should keep commits from master until 2016-06-10 (https://github.com/KiCad/kicad-doc/commit/827eb6dd81a40ba2675f67e06823e23b87907858),
since after these commits progressive reconsturuction changes had started.

And this will save these 4.0 translations.
- Catalan
- French
- German
- Italian
- Japanese
- Russian 
- Spanish

I tested in my local PC that this could be done by git merge from master to 4.0 and git rebase -i fbdc596b99.

kinichiro, 2016-07-12:
This diagram is not needed, though, I would like post this comment to share the image of separation.
----------+----------+----------+----------+----------+----------+----------+----------+

Now, all of writers and translators are working on `<master>` branch only.

```
<master>  [+]---------------------+   -  -  -  - {writers}{translators}{web}{pkg}
           |
           V
<stable>  [#]
          <4.0>
```

We should separate the branch to work.

```
<master>  [+]--------------------[+]  -  -  -  - {writers}
           |                      |
           V                      V
<stable>  [#]------$-----        [#]  -  -  -  - {translators}{web}{pkg}
          <4.0>    4.0.3         <5.0>
```

1. branch `<master>` for writers to update original English adoc, this is doc nightly-build branch.
2. branch `<stable>(e.g. 4.0 or 5.0)` for translators work, this is doc stable branch.
3. when the `new <stable>` branch is created, copy po files from `older <stable>` branch. (optional)

starfort-jp, 2016-07-12:
Hello, everyone.

I think that "separating branches for writers and translators" is one good idea.
Current translation of documents on KiCad Website are incomplete even though the translation is done. Whenever writers make changes, it comes out. Even correcting a typo...

I have one question. Why not use .POT files?
My understanding of typical "Translate With GetText" is as follows.
 1) Writers produce .pot files. (This file includes only the original texts, which need translation.)
 2) Translators create/update .po files from .pot files and send it back.
In this case, no synchronization required. Separate it or not, wouldn't it be better？

nickoe, 2016-07-12:
It has always been the idea that translations are supposed to be complete on the stable branches, and not on the master branch, mainly. This has just not been fulfilled in real life, probably mostly because of no process documentation for this method. I received little or no comments on the initial docs versioning description that I linked earlier.

The reason for the 4.0.0, 4.0.1, 4.0.2 being equal, is partly because the project lead did not want to update the docs for the subsequent 4.0 releases and because there was no proper process used for the docs and i18n repos.

The principle about versioning them as the kicad version triplet and an appended integer to the version for a "snapshot" of a doc, is probably a bit too complex and not needed, now that we aligned with the project lead to have a grace period between the source being tagged and we announce the real release packages.

nickoe, 2016-07-12:
@kinichiro I have performed the rebase you suggested and pushed it to the 4.0 branch. Please verify.

starfort-jp, 2016-07-12:
POT file and PO file are basically the same. Changing a file extension only.

I use .pot file in this project. You can see .pot file includes the original texts only, but essentially the same as .po file.   
https://github.com/starfort-jp/KiCAD_ComponentUtility/tree/master/locale

Once .pot file is created, translators can use "Poedit" to create/update .po files from .pot files whenever he/she likes. 
[image]

I'm really sorry, I don't know much about cmake. (I'm a Pascal user.)

@nickoe  FYI. 
https://www.icanlocalize.com/site/tutorials/how-to-translate-with-gettext-po-and-pot-files/
http://stackoverflow.com/questions/2694742/is-there-a-cmake-function-to-update-pot-files

kinichiro, 2016-07-12:
@nickoe thanks.
Could you add the commits on Jun 10, 2016 too, please ?
That would save cvpcb transportation of ru.

kinichiro, 2016-07-13:
@starfort-jp
I have a question about .pot and asciidoc.
To generate translated doc, kicad-doc is using po4a tool.
It substitutes original English sentence from English.adoc, 
then find translated sentence from Localized.po,
and creates Localized.adoc.
After that, asciidoc processes Localized.adoc to html or pdf.

I don't think po4a considers .pot file while its processing.
In case it does, there is no way to link .pot and modified English.adoc, I believe.

Do I miss anything ?

starfort-jp, 2016-07-13:
@kinichiro 
I read po4a's man page.
http://po4a.alioth.debian.org/man/man7/po4a.7.php
http://po4a.alioth.debian.org/man/man1/po4a.1.php

It shows po4a is supposed to use with .pot file. (Please refer to "Graphical overview" in man7.)
Also mentioned, "this tool cannot be used to convert existing translations to the po4a system. For that task, please refer to po4a-gettextize" in man1.

My understanding is as follows.
 1. To create .pot file, use "po4a-gettextize."
 2. Don't use "po4a-updatepo", use "Poedit" insted of it.
 3. To convert from .po file to doc file, use "po4a-translate."

Sorry, I don't use Linux. (I can't use po4a tools.)
So I can't confirm the compatibility between "po4a-updatepo" and "Poedit."

If my understanding is wrong, please let me know. 
Thank you.

starfort-jp, 2016-07-13:
Now I've found the incompatibility.
It seems that Poedit is inferior to po4a-updatepo in detection of same strings.
I realize the reason not to use .pot files here.

I'm hoping for the separating branches solution.

kinichiro, 2016-07-13:
@starfort-jp
I see. But, thanks for your information.
That helped us to taking consideration for every methods to choose.
I would like to progress the separate branch method.

kinichiro, 2016-07-13:
> The overall procedure just needs to be documented, in relation to lp:kicad, kicad-doc and kicad-i18n.

I'll try to think of above, and I would like to mention explicitly
"writers should edit on `master`, translators should work on stable branch (e.g. `4.0`)."

I am going to read these current instructions again, and will submit PR to `master` branch.
- README.adoc
- docs-versioning.adoc
- AUTHORS_README.adoc
- translation_instructions.adoc

nickoe, 2016-07-14:
@kinichiro OK, I also cherry-picked 5d99e80f36cf2050ee93f753202996fad92f878c onto 4.0.

kinichiro, 2016-07-15:
@nickoe Thanks for cherry-picking. I checked it.

I tried to write down relations between branches etc. by PR #411 ,
and any comments would be highly appreciated.

And, I have a question.
To put the stable `4.0` docs on the [KiCad Web site docs area](http://kicad-pcb.org/help/documentation/ ), what is required ?
Currently, I believe it publishes development `master` docs, and I want KiCad Web site to publish stable one.

nickoe, 2016-07-15:
@kinichiro First that I make it such that the stable versions are hosted on the docs.kicad-pcb.org.  I have now done that. Secondly, we just need to link the docs from the web site. I don't know how we can present that best. Do you want it to link the latest stable or all versions? I guess the most sane is latest stable maybe? But then we might need to remove some languages, because they might not be available on the stable compared to master.

kinichiro, 2016-07-16:
@nickoe  I believe, 

- web site should mainly provide latest stable (currently 4.0) docs. And if new translated version is commited in 4.0, it should be reflected to the web site.
- minor versions of stable (4.0.0, 4.0.1, etc.) not needed on the web site.
- if we provides past historical versions (bzr4022 etc), another page, say "Old Docs" or something, should serve them.
- I don't think we don't have to provide master branch docs on the web site, since those are under development and able to read them from Github.
- If some languages are not included in the stable 4.0 branch, but included in master dev branch or past historical version, we cannot provide them as the stable docs. But, "Old Docs" page or directory read from Github could save those languages.

BTW, I thought there is no language that is in master branch, but not in stable 4.0 branch.

nickoe, 2016-07-16:
@kinichiro Are you going to update the translations for pcbnew and getting_started_in_kicad on the 4.0 banch for jp soon? According to the translation status update script they are still missing for jp.

kinichiro, 2016-07-17:
@nickoe Yes, we will update Japanese translations on 4.0 branch for 4.0.3
release.

kinichiro, 2016-07-17:
@nickoe ja translations on 4.0 branch for release 4.0.3 has done.
And results executing utils/status/TRANSLATION_STATUS_UPDATE.sh here.

```
cvpcb, ja, 261 translated
eeschema, ja, 1417 translated
gerbview, ja, 115 translated
getting_started_in_kicad, ja, 343 translated
idf_exporter, ja, 98 translated
kicad, ja, 172 translated
pcbnew, ja, 1312 translated
pl_editor, ja, 241 translated
```

kinichiro, 2016-07-22:
About docs version and KiCad Web.

Currently, structure of http://docs.kicad-pcb.org/ is like this.

```
Index of /
----------
../
4.0.0/
4.0.1/
4.0.2/
ca/   
de/   
en/   
es/   
fr/   
it/   
ja/   
nl/   
pl/   
ru/   
----------
```

**Status**
3 directories 4.0.0/, 4.0.1/ and 4.0.2/ are maybe built from stable `4.0` branch by tag.
And these contains 6 languages (en, fr, it, ja, nl and pl), now.
4 languages (ca, de, es and ru) are not included here since those translations were added after 4.0.2 tag (10-DEC-2015).

10 language directories are maybe built from `master` branch.
And these docs are linked from KiCad Web ( http://kicad-pcb.org/help/documentation/ ), now.

Soon, 4.0.3 tag will be put on the stable `4.0` branch after some translation period,
and directory 4.0.3/ will be created.
I think this 4.0.3/ docs should be linked from KiCad Web as latest stable `4.0` docs.

**Suggestions**
BTW, in this procedure, KiCad Web have to update docs link URL for every minor update since directory name will be changed.
I would like to suggest creating `stable/` directory and put latest stable docs into there.
Next minor update time, rename `stable/` to `4.0.3/` and recreate `stable/` again.
Also, `master/` directory should contain above 10 lang dirs, then we can announce latest nightly build docs are referenced from http://docs.kicad-pcb.org/master/ .

```
Index of /
----------
../
4.0.0/          # <- save old version docs
4.0.1/          #
4.0.2/          #
master/         # <- latest nightly docs
stable/         # <- stable docs
----------
```

These changes will affect build procedure and scripts, and I think it is not easy to modify, though, I appreciate if taking consider these suggestions.

Thanks.

nickoe, 2016-07-22:
Yeah, that is becasue I had some trouble setting up the server in the beginning. Thank you for reminding me again. I should be done now. It is currently as follows:
```
../
4.0.0/
4.0.1/
4.0.2/
master/
stable/
kicad-doc-HEAD.tar.gz
kicad-doc-HEAD.zip
```
I have kept the tar.gz and zip files in case some auto builders depends on those, at least for now.

nickoe, 2016-07-22:
@kinichiro But this also means that we have to do something different for the links on the kicad website for the docs that is linked now in languages that are not supported in the stable. What to do with those? (Right now they are just not found)

kinichiro, 2016-07-23:
@nickoe Thanks, I saw docs on the web site are linked to the stable.
And I would like to suggest that we put docs built from HEAD of the `4.0`
branch without any tags into the stable directory.
Then we can rescue ca, de and ru, those languages are not found now.
The docs from tag 4.0.2 are kept in directory 4.0.2 already, so we can
store the latest stable docs into the `stable` directory.
Is this acceptable or not ?
Thanks.

nickoe, 2016-07-24:
Yes, I have not made the HEAD of the 4.0 branch be in the stable directory.

kinichiro, 2016-07-24:
@nickoe 
Many thanks to your great works. I could verify every translated latest stable version 4.0 of docs were on the KiCad web site.

ciampix, 2016-08-24:
I thank you all, you did a great work and I just sat lurking your interesting suggestions.
But now I really would like to move this debate on the doc dev ml on launchpad...

kinichiro, 2016-08-24:
I agree.
Next time if I have a topic that should be talked by the team, I will post
that topic to doc ML.

I think the topic of this issue was solved, and I will close this in a few
days.

closed 2016-12-10 by kinichiro