KiCad/kicad-doc #415: File format documentation

Tim-EE, 2016-08-24:
There doesn't appear to be any up to date documentation (in a single place) for the KiCAD file formats.

Is there any desire to document the various  file formats used in KiCAD here?

e.g. change the documents at: http://bazaar.launchpad.net/~stambaughw/kicad/doc-read-only/files/head:/doc/help/file_formats/docs_src/ to AsciiDoc and put them in this repo.

ciampix, 2016-08-25:
IMHO we should ask to devs since those are devs docs not users docs.

Tim-EE, 2016-08-25:
Personally, as a user, I have used them as I find it easier to layout a footprint using the new *.kicad_mod format, but I see your point.

The argument to put it in either repo could be made.

Perhaps we should ask @stambaughw what he thinks, they are his documents after all?

stambaughw, 2016-08-25:
Yes, we should create these documents.  I'm not terribly concerned about
documenting the legacy file formats at this time.  The current legacy
file formats are woefully outdated.  I am speaking from first hand
experience while writing the new schematic I/O plugin.  That being said,
I have no problem with someone creating the documentation for the legacy
file formats should they feel motivated.  Please note that the new
schematic and symbol library file formats should be done around the end
of October if all goes as planned so those file formats will change.

As to where we should put these documents, I'm leaning towards the
developers documentation as opposed to the user's documentation.  I
don't think the majority of users will need the file format
documentation.  I typically find that most users who actually use the
file format documentation are also developers so they do not fit your
typical user profile.

johnbeard, 2017-04-04:
How would these be maintained in the main codebase? Using doxygen-markdown for the format documents would be very painful compared to asciidoc, and much less translateable.

nickoe, 2017-11-09:
@stambaughw Where are the latest docs if someone is going to convert them to adoc?  And should we keep them in the kicad repo as part of the doxygen docs or should we include it in the kicad-doc repo or maybe even embed them on the website? I think kicad-doc would be most suitable.

stambaughw, 2017-11-09:
Nick,

The original doc backup before the conversion to adoc can be found at
https://code.launchpad.net/~stambaughw/kicad/doc-read-only.  The file
format docs are in the doc/help/file_formats/folder.  I don't know if
anyone maintains a separate branch with more recent changes.  I suspect
not but you never know so it might be worthwhile asking on the dev
mailing list.

Cheers,

Wayne

iromero91, 2017-12-25:
If someone would like my two cents, the file format documentation should be in the kicad source tree (Doxygen, adoc, etc). And any commit that makes a file format change, apart from changing the file version number it also *MUST* update the documentation, and commits that change the former without the latter should be rejected by the project. Keeping up to date documentation is very important for interoperability (even with KiCAD itself!) It's just a tiny amount of extra work every time something is changed but file format changes are no light matter anyway!

ciampix, 2018-02-08:
Agree with @iromero91 for the update issue. But I have to say that we agreed to branch docs with KiCad versions and to have similar versioning numbers so, if devels behave correctly we should have format docs aligned with KiCad versions as it should be with the user docs...