KiCad/kicad-doc #416: Difficult to zoom in HTML docs on Firefox Mobile because of sidebar

w1res, 2016-08-30:
I'm trying to read the HTML version of the docs on Firefox Mobile on Android (version 48.0) and having issues with the text size.

The text was very small, so normally I would zoom in and the text would get larger; however, when I try to zoom, the sidebar gets larger as well, and this causes the sidebar to overlap the text, making it difficult to read.

The desired behavior would be to zoom in just on the text, or the whole viewport, so that the sidebar doesn't start to encroach over the article text.

As a side note, Firefox has a Reading Mode that strips away the navigation and styling, making things easy to read on mobile. This doesn't work either and the order of the articles gets mixed up.

w1res, 2016-08-30:
For example in Reading Mode on [this](http://docs.kicad-pcb.org/stable/en/kicad.html), it starts at section 2.1 instead of section 1.

ciampix, 2016-08-30:
Ah I see. I'll investigate if there is a way to turn on / off the lateral bar. A simple way is to build the HTML with TOC on the top.

HiGregSmith, 2017-08-31:
This affects iPhone too.