KiCad/kicad-doc #419: Global Spread and Place: Only for default canvas

zz-1b, 2016-09-11:
Chapter 7.4 Automatic Footprint Distribution explains how to have parts placed automatically through a command in the footprint context menu.
However, the "Global Spread and Place" command is only available if the canvas is switched to 'default' (via F9), not if it is in OpenGL or Cairo mode.

It might save others lots of confusion and wasted time to mention that dependency in the documentation.

nickoe, 2016-09-11:
That could possible be made as a note in the doc.

keruseykaryu, 2017-07-22:
The full automatic spread is available in OpenGL mode only in nightlies. It works without special user action: when you read netlist the componets are spreaded and selected.
This alternative behavior has been mentioned in chapter 4.4.3.

grbd, 2018-05-01:
Within the Global spread and place menu there are also options for
Automatically placing all footprints, which is different from just spreading the footprints.

The auto place feature, allows you to lock / unlock certain footprints in position (like connectors), then autoplace the rest within the edge cut boundary