KiCad/kicad-doc #438: Python sample code does not work

kinichiro, 2016-10-24:
I found python sample code in the Pcbnew manual chapter 15 does not work well.

- 15.3 : GetValueObj() and GetReferenceObj() are invalid. Value() and Reference() are OK.
- 15.4 : This sample should be rewrite for .kicad_mod not for old .mod.
- 15.5 : SEGVIA should be VIA. pcb.GetSegZones() is invalid.
- 15.6 : not tested yet.
- 15.7 : not tested yet.

ciampix, 2016-10-27:
Many part of the manual(s) must be reviewed. We should open a bug report for every problem detected in an (as more as possible) atomic way. This helps fix the translations and the bug report revision. Thanks for your contribution, it is greatly appreciated.

davidbrenner, 2016-12-07:
I ran into this problem today and spent a couple hours working through all the examples. I wish I would have realized earlier that there was a kicad-doc repo as I would have edited while working through the examples and submitted pull requests for my changes. I will still do that at some point this week. When I do, I'll open an issue for each of these problems and comment back here. I've tested all the examples except 15.4 (I don't have any of the legacy .mod libraries on my system) and 15.7.

In case anyone else gets to this before I do, the examples at pcbnew/python/examples are helpful for sorting through the changes to the "C++ API". I also extensively used git log's search feature (git log -S) to look through the commit history to see where/when these changes happened as I'm not yet familiar with KiCAD's codebase. I also found the doxygen C++ API reference somewhat helpful. Overall, this was actually a decent introduction to KiCAD's code base (though I still certainly have a lot to learn about it).

davidbrenner, 2016-12-08:
Created a separate issue and pull request for each code example to fix the problems.

Is it possible to link the code examples directly from the KiCAD kicad-git/src/kicad-git/pcbnew/python/examples directory? That would prevent this issue from recurring again.

nickoe, 2016-12-08:
@davidbrenner Thank you for following up on this. I have merged your pull requests.

I don't think we can easily link the code directly. Is there a chance that you could try to test it for the stable branch too against 4.0.5, and maybe fix them there such that we can have it updated for the 4.0.5 of the docs?

davidbrenner, 2016-12-08:
Unfortunately, I don't have the time to test that right now. I did a quick diff on the pcbnew/python/examples directory and don't see any changes. Assuming I didn't screw that up, I think it's safe to say it is compatible for v4.0.5 of the docs.

davidbrenner, 2016-12-08:
Eh, I may have screwed up that diff.. after double checking myself, with v4.0.5 I don't see the examples directory now.

kinichiro, 2016-12-09:
Hi,

I tested updated python script samples by @davidbrenner
on Ubuntu16.04 with KiCad 4.0.4+e1-6308~48~ubuntu16.04.1 amd64.

I could confirm these 3 samples below worked ok.

- 15.3 Example that loads a board, hides all values, shows all references
- 15.5 Board Inspection Example
- 15.6 We only want to change pins from 1 to 14, 15 is a thermal pad that must be kept as it is.

I tested these samples like this.

```
$ ./sample.py tutorial1.kicad_pcb
```

But, I could not verify this sample running fine.

- 15.7 Build footprints easily filling in parameters.

I tested these samples like this.

```
$ export PYTHONPATH=/usr/share/kicad/scripting/plugins
$ ./sample.py
```

15.7 sample python had just finished without any output.
Has anyone tested this sample ?

And if these samples are OK, I would like to cherry-pick those 4 commits from master to 4.0 docs.
These commits changes source code only, then cherry-picking  will not invalidate translations.

Thanks.

nickoe, 2016-12-09:
Ok, thanks. I will probaly recheck that during the weekend. I would also
like to verify them against 4.0.5.

kinichiro, 2016-12-10:
I found some python samples come from source code tree like below.

- 15.3 : pcbnew/scripting/examples/hidePcbValuesShowReferences.py
- 15.4 : pcbnew/scripting/examples/listPcbLibrary.py
- 15.5 : pcbnew/scripting/examples/listPcb.py
- 15.6 : none
- 15.7 : pcbnew/scripting/plugins/FPC_(SMD_type)_footprintwizard.py

Now I'm trying to inspect 15.7 script by python debugger (pdb).

kinichiro, 2016-12-10:
I posted PR to fix 15.4 sample https://github.com/KiCad/kicad-doc/pull/457 .
This python script was provided by JP.
Thanks a lot, JP.
I confirmed this python script running successfully on KiCad 4.0.4.

I also sent PR to fix python description https://github.com/KiCad/kicad-doc/pull/458 .
This fixes 1) SEGVIA -> VIA and 2) remove GetSegZones().

These 2 PRs are targeted for master branch.
I separated these 2 fixes into 2 PR so that we can choose applying these changes to 4.0 doc or not.

kinichiro, 2016-12-11:
I could work with 15.7 sample python wizard script.
First of all, I had misunderstood the way to use of this python code.
I should open the Footprint editor first, then press top toolbar button 
"New footprint using the footprint wizard", and press top toolbar button
"Select the wizard script to load and run".
After this operations, I could see the list of wizard python script under
/usr/share/kicad/scripting/plugins/ or C:\Program Files\KiCad\share\kicad\scripting\plugins .

But the sample code fixed by #456 didn't work and got error.
I noticed that #456 code was differ from /usr/share/kicad/scripting/plugins/FPC_(SMD_type)_footprintwizard.py a little bit.
Then, I just copied wizard code to the manual by PR #460.

kinichiro, 2016-12-11:
I would like to close this issue for 4.0.5 release in 2 or 3 days.
I will do these jobs tomorrow.

- Merge PR #457 and #458 to master branch
- Cherry-pick from PR #449, #452, #454, #456, #457, #458 and #460 to 4.0 branch

These PRs change Pcbnew manual only.
Please feel free to comment for this.

nickoe, 2016-12-11:
@kinichiro I merged #457 becasuse it looks good. The other #458, I am a little bit in doubt of the place to put the scripts on windows. Needs to be verified.

kinichiro, 2016-12-12:
@nickoe thanks.
I posted PR #461 for 4.0 branch and I tested to build whole 4.0 docs in my local.

closed 2016-12-14 by kinichiro