KiCad/kicad-doc #440: GSIK not displaying example graphics

ClarkInAz, 2016-11-04:
I'd started with KiCad a few years ago, but got "distracted" by a work requirement to [be ab]use[d by] Buzzard, er, Eagle.  That's now passed, so thought I'd try KiCad (since I'm dissatisfied with trying to do a PCB under Fritzing, and find Buzzard, er, Eagle to be counter-intuitive).

Problem:  Going through GSIK, section 3.1 step 24 says "Organise all components on your schematic sheet as shown below."  The next line is (what appears to be) the text "gsik_totorial1_010_png", though I can't highlight the text (when I try to, it "moves" rather than highlights).  (I strongly suspect that the intent is to display a picture here.)

Versions being used:
KiCad 4.0.4-stable
OSX 10.11.6 on a Mac Book Pro
Firefox 49.0.2

kinichiro, 2016-11-04:
Hi,
Could you try to see GSiK on the web from here ?
http://docs.kicad-pcb.org/stable/en/getting_started_in_kicad.html

Now I have no idea why you cannot see the image on your local environment,
I would like to see it.

Sorry for inconvenient !

ClarkInAz, 2016-11-05:
OK, I just tried the indicated link -- I can see the graphics in it.  (IMHO, there's almost too much graphics there, but sometimes "too much is better than too little".)  I set a bookmark in Firefox!  :-)

I grabbed a "screen shot" showing the problem.  I hope I'll be able to put them here...
[image]

nickoe, 2016-12-11:
Related: https://bugs.launchpad.net/kicad/+bug/1628550

kinichiro, 2016-12-11:
Thanks for informing that.
This seems OSX packaging issue and already resolved in nightly build
package.
Let's see 4.0.5 package.

nickoe, 2016-12-12:
A guy on IRC tried to test it, and it seems that the link contains a trailing slash which apparently breaks this. It is not easy to see in the screen shot, but I guess this might be a subtle bug with how the URL is constructed in KiCad.  @ClarkInAz Can you verify that?

ciampix, 2016-12-12:
I do not see any traliling slash in the sources...

marco@lap3-ciampix:~/git/github/kicad/kicad-doc/src/getting_started_in_kicad$ grep gsik_tutorial1_ *
getting_started_in_kicad.adoc:image::images/gsik_tutorial1_010.png[gsik_tutorial1_010_png]
getting_started_in_kicad.adoc:image::images/gsik_tutorial1_020.png[gsik_tutorial1_020_png]
getting_started_in_kicad.adoc:image::images/gsik_tutorial1_030.png[gsik_tutorial1_030_png]
getting_started_in_kicad.adoc:image::images/gsik_tutorial1_040.png[gsik_tutorial1_040_png]
getting_started_in_kicad.adoc:image::images/gsik_tutorial1_050.png[gsik_tutorial1_050_png]
getting_started_in_kicad.adoc:image::images/gsik_tutorial1_060.png[gsik_tutorial1_060_png]
getting_started_in_kicad.adoc:image::images/gsik_tutorial1_070.png[gsik_tutorial1_070_png]
getting_started_in_kicad.adoc:image::images/gsik_tutorial1_080.png[gsik_tutorial1_080_png]
getting_started_in_kicad.adoc:image::images/gsik_tutorial1_090.png[gsik_tutorial1_090_png]
getting_started_in_kicad.adoc:image::images/gsik_tutorial1_100.png[gsik_tutorial1_100_png]
grep: images: Is a directory
grep: po: Is a directory

nickoe, 2016-12-12:
It is not for the image link, but for the html link directly.

cvgaviao, 2016-12-21:
I'm facing this problem with in nightly build at ubuntu also. 
I noted that all images (*.png) are located in the "/usr/share/doc/kicad/help/en" but the html files are expecting them to be in "/usr/share/doc/kicad/help/en/images"

nickoe, 2017-03-04:
It looks like something we can't fix in the docs and the op did not respond for a long time.

closed 2017-03-04 by nickoe