KiCad/kicad-doc #462: Call for the translation for 4.0.5

kinichiro, 2016-12-12:
Hi Translators,

Could you update translation for Pcbnew manual for 4.0 branch by **Dec 16th**, please ?
For the upcoming KiCad 4.0.5 release, we fixed python section of Pcbnew.
Take a look at this PR for changes.
https://github.com/KiCad/kicad-doc/pull/461

There are 6 po files under the pcbnew/po/ directory.
de.po  es.po  fr.po  it.po  ja.po  pl.po

- Please reply here **your language** if you can work on the translation work
- Post your PR to **4.0** branch of GitHub until **Dec 16th**, Friday, please.
- If no one raise hand for translation, we will apply changes to that .po file.

Sure, other 4.0 doc translation updates are welcomed, too.

Best regards,

This issue will be close after this translation work period.

ciampix, 2016-12-13:
My work on the pcbnew manual translation in Italian is not even started. I barely managed to finish the translation of Eeschema manual just a few days ago. I can update the Italian po files or let you do it. I'll do what you prefer / you think is better.
TIA

kinichiro, 2016-12-13:
@ciampix Thanks for informing your situations.
I understand and take your time as you can.
I will not update Italian po files for Pcbnew manual now.

kinichiro, 2016-12-13:
I will maintain FR po file.

kinichiro, 2016-12-14:
Current status.

- de : planned.
- es : finished by @ElektroQuark. Thanks.
- fr : PR #467 and merged.
- it : translation is ongoing and I will leave this as it is.
- ja : PR #469 and merged.
- pl : won't update since chapter 15 was not translated yet.

French Pcbnew translation still has 39 fuzzy and 3 untranslated.
I appreciate it if someone would complete the rest of French translations.

keruseykaryu, 2016-12-14:
"pl : won't update since chapter 15 was not translated yet."
It has been translated in master branch. One can take translations from master branch and transfer to the 4.0 branch.

kinichiro, 2016-12-14:
@keruseykaryu 
Thanks for heads up.
Now I noticed that pl.po from `master` has only 25 fuzzy and 33 untranslated
if I copied it to `4.0` branch.
May I copy pl.po on `master` to `4.0` stable branch ?

kinichiro, 2016-12-14:
@keruseykaryu 
I compared all pl.po between master and 4.0.

- For `idf_exporter` and `pl_editor`, copying pl.po from master branch to 4.0 branch resulted complete 100% translation.
- For `eeschema`, 62 untranslated was reduced to 6.
- For `cvpcb`, `gerbview`, `GSiK` and 'kicad', invalidate of translation increase.

Then I thought copying pl.po of `pcbnew`, `eeschema`, `idf_exporter` and `pl_editor` to **4.0** from **master** brings better translation status for PL.
How do you like this ?

Sorry but I can't tell which translations are better between master and 4.0,
so this proposal comes from just the number of fuzzy and untranslated.

keruseykaryu, 2016-12-14:
@kinichiro 
I think you should ask the translation leader(s).

Anyway, someone should re-check these translations.
I can't, because I'm not a "Linux guy" - IMO KiCad has become a _very scripted thing_, so there are no place for "Redmond guys"; and I don't want because I said "Enough!" to myself. Sorry.

kinichiro, 2016-12-14:
I checked the commit history of pl.po on both master and 4.0.

- Translations of Pcbnew, idf_exporter, pl_editor are updated on master
branch, then I would like to copy them to 4.0.
- Eeschema was updated on 4.0 recently, then I would keep it as curretly.

@ciampix How do you like this ?

ciampix, 2016-12-15:
First of all @kinichiro, thanks for your beautiful work you are doing on the translation... 
I have strove to keep the pl translation updated, merging the master and 4 branch if possible. Then you should find the pl in a fairly good shape. When it was not possible, I gave up keeping the two branches in sync since they diverged too much. @keruseykaryu used to update the master branch _before_ we decided to create the actual organization. Also thanks to @keruseykaryu to make me think on the kind of problems translators could face when master docs are updated that I think this document organization should minimize from now on. Now I think it is better to manually update all the pl.po's since @keruseykaryu is not able to do it by himself not having a doc building environment working in its place. Then I would ask @keruseykary if he can translate _just the 4.0 branch_ for the (I hope) few missing strings. TIA to all!

kinichiro, 2016-12-15:
@ciampix I understand the situation of Polish translations. Thanks.
I will wait and do nothing for pl, now.

@keruseykaryu, if you can translate and you need some help, feel free to
say it to us, please.

IMHO,
I think copying some pl.po from master to 4.0 branch would reduce the work
of manual translation job.

Best regards,

kinichiro, 2016-12-15:
Current status.

- de : finished by @tijuca . Thanks.
- es : finished by @ElektroQuark . Thanks.
- fr : PR #467 and merged.
- it : translation is ongoing and I will leave this as it is.
- ja : PR #469 and merged.
- pl : leave as it is.

I think translation work for 4.0.5 is completed already.
Is there some updates ?

ciampix, 2016-12-15:
For my part no. I will update in the future but you can consider "it" frozen and ready to be packaged. TIA

keruseykaryu, 2016-12-15:
Don't know if it will be good.
[pcbnew-po_pl.zip](https://github.com/KiCad/kicad-doc/files/655489/pcbnew-po_pl.zip)

nickoe, 2016-12-15:
@keruseykaryu yeah, that looks good to me. Looks like 100% translation for pcbnew to polish. Also, it might be better use of a translators time to only trach the stable branch and not the master branch. Should I commit it with you as the author, @keruseykaryu?

kinichiro, 2016-12-16:
@keruseykaryu 's translation update is great.
I think addendum.pl should also be updated from 2015 to 2016.

keruseykaryu, 2016-12-16:
@nickoe @kinichiro Do what you think is appropriate.

nickoe, 2016-12-16:
I committed the change, but I clearly forgot to build, only ran the status script. Sorry for the inconvience, I will try to fix that when I get home unless someone beats me to it.

kinichiro, 2016-12-16:
I added some commits for Pcbnew Polish after @keruseykaryu with @nickoe.

- update addendum.pl https://github.com/KiCad/kicad-doc/commit/95e75f331260b18eb1ce3a415c8d668d4c72332c
- copy updated screenshots from master to 4.0 https://github.com/KiCad/kicad-doc/commit/a5052454a8458c560b3df681c3cd4aec4ec478f7

After these 2 commits, I could verify that build on 4.0 branch was succeeded.

If translator uses only Windows and want someone to updatepo, we can support that.
Even if translator does not have asciidoc build environment, submitting PR causes Travis-ci automated build.
We are trying to keep translations sane on `stable 4.0 branch` as far as we can.
Feel free to ask every support, please.

Thanks

keruseykaryu, 2016-12-16:
Ok, people. It's Christmas Time. Remaining translations for 4.0.
[src_pl.zip](https://github.com/KiCad/kicad-doc/files/657713/src_pl.zip)

kinichiro, 2016-12-16:
@keruseykaryu Thanks, I will check your translation for 4.0 branch update soon.

kinichiro, 2016-12-16:
@keruseykaryu Do you have src/eeschema/images/pl/eeschema_hierarchical_label.png ?
Only this screenshot image is missing.
If you can, give me that file.

kinichiro, 2016-12-16:
I posted PR #471 and it includes English screenshots for eeschema_hierarchical_label.png instead of Polish one. If I could get Polish screenshots, I will replace it. In my local environment, I could build all 7 updated Polish translations, and now Travis-CI is running to exam whole build.

keruseykaryu, 2016-12-16:
I don't have this file (Don't know why), but I captured new one according to English one. 
The only problem is I've Win10 now and I have some pixel work to replace frame as Win8 to match other screenshots ;-). Hold on second.

keruseykaryu, 2016-12-16:
Here it is.
[eeschema_hierarchical_label.zip](https://github.com/KiCad/kicad-doc/files/657932/eeschema_hierarchical_label.zip)

kinichiro, 2016-12-16:
Thanks a lot ! I could push your new screenshot into PR #471 .

kinichiro, 2016-12-17:
Translation period for 4.0.5 is finished, and this issue will be closed soon.
Are there any issues for this ?
Thanks for all translators :-)

nickoe, 2016-12-17:
I think we are done. I have tagged it now.

closed 2016-12-17 by nickoe