KiCad/kicad-doc #501: Equivalent python doxygen doc for 4.0.7 Stable?

HiGregSmith, 2017-08-31:
Is there an equivalent to 

http://docs.kicad-pcb.org/doxygen-python/

For 4.0.7 Stable?

ciampix, 2017-09-03:
Sorry I do not understand the question. I do not see any content at that URL

HiGregSmith, 2017-09-03:
Clicking on "Namespaces" tab and then the "pcbnew" label brings you to this page:
http://docs.kicad-pcb.org/doxygen-python/namespacepcbnew.html

ciampix, 2017-09-04:
Ok you probably have to write to the dev mailing list since the program api documentation are under dev control. API are part of the KiCad source repo and are not part of the KiCad doc project repo nor the doc  maintainers have any access to it. I even do not have write access to the source repo myself!

ciampix, 2019-03-25:
More than 1 year old and no reply... I'm closing this...

closed 2019-03-25 by ciampix

nickoe, 2019-03-26:
Tracked in https://github.com/KiCad/kicad-doc-website/issues/14