KiCad/kicad-doc #507: Getting_started_in_Kicad : Default FootPrint libraries + footprint_list_documentation icon

laurentc2, 2017-10-01:
I just downloaded KiCad 4.0.7 and installed it on Windows8-64 without connection to internet.
Then I have followed the tutorial of the doc directory : **getting_started_in_kicad** 
At the steps 49-50, "_Run Cvpcb_", I got a problem, because Kicad should load many footprint libraries from internet:
> I think that those default libraries loaded at the first Cvpcb should be included in the default installation package, not afterwards.
At the step 52, I got another problem : it says that R foot print should be : _Discret:R1_
but that _discret_ footprint library does not exist.
At the step 53, I should click on the _Display footprint list documentation_ icon to get a multi-page PDF document with all available footprints : that icon does not exist !

Thank you to correct those issues either in the doc, or better in the Kicad distribution for the future readers.
Laurent

poeschlr, 2017-10-02:
I can not give input on documentation but i can tell you that we librarians try to get the developers to drop the current github plugin implementation for something that makes more sense.
See mailing list: https://lists.launchpad.net/kicad-developers/msg30892.html

To clear things up: Cvpcb will download zip archives of the footprint libs the first time you start it in each session. (If you close kicad and reopen it, cvpcb will again download all libs!)
If you want local libs might i suggest this great tutorial by @bobc:
https://github.com/bobc/kicad-doc/wiki/How-to-install-local-Footprint-libraries

Having said that, i don't think it is a good idea to include the libs into the installer. (The 3d lib nears 1GB of download size even when compressed.)
So it will (In the future) be separated in some way from the core package. (Ubuntu already has a separate lib package.) Ideally the user can update all libs at the same time from within kicad. But there is as of yet no clear idea of how it will look like.

EeliK, 2018-06-11:
In KiCad v5 the library internet loading problem will indeed be away.

I have fixed the 'discret' problem for v5.

I have removed the documentation list from 53, it doesn't exist as was noticed.

All in all, these all are fixed for v5.