KiCad/kicad-doc #537: Italian translation wrong on make-schematic-components-in-kicad ?

Ozzyboshi, 2018-02-16:
At http://docs.kicad-pcb.org/stable/it/getting_started_in_kicad.html#make-schematic-components-in-kicad I read 

"Nella finestra delle proprietà del pin che appare, impostare il nome del pin a VCC, impostare il numero del pin a 1, e il Funzionalità elettrica a **Uscita alimentazione** poi fare clic su OK."

In the same document, but english version at http://docs.kicad-pcb.org/stable/en/getting_started_in_kicad.html#make-schematic-components-in-kicad
same sentence:

In the Pin Properties window that appears, set the pin name to VCC, set the pin number to 1, and the Electrical type to **Passive** then click OK.

Which electrical type is correct? Power output as stated in the italian version or passive as stated in the english version?

The underneath image does not help because it shows "power input" as electrical type.
I assume that the wrong version is the italian one, but I am a beginner so don't take my word for it.
Thanks

ciampix, 2018-02-16:
Good point... but I have to say that being alone as the only Italian translator of the manuals I have translated in a rush all "stable" docs and now I am focusing on the "master" (i.e. future v5 of KiCad that it is expected to be released very soon) of the docs. Having said this I observe that even the English authors seem to be more concentrated on the future version since the master version is different, and this time the translation is correct, see:

http://docs.kicad-pcb.org/master/en/getting_started_in_kicad.html#make-schematic-components-in-kicad

http://docs.kicad-pcb.org/master/it/getting_started_in_kicad.html#make-schematic-components-in-kicad

Probably the "master" version is the correct one and I have to confess I didn't check it. Will you please confirm this? I can fix both versions if you think it is worth...

Ozzyboshi, 2018-02-18:
The english master version makes sense, it states:

Pin 1: Power input
Pin 2: Passive
Pin 3: Passive

That I think reflects the puropose of the  custom component.
The italian master version is somewhat weird, it lists pin 2 as "ingresso alimentazione" and pin 3 as "Uscita alimentazione".
I think the master english version is the correct one but, again, I am a total beginner so dont' trust my judgement.

poeschlr, 2018-02-18:
That whole tutorial does not really make sense. It might be better to either state that the electrical type for the purpose of this tutorial can be set to passive. You can add a link to the section explaining electrical types for users who want to dive into this part.

Or you would need to first define what component you want to make. In particular you need to state why you set a pin to a specific electrical type.
(One option would be a screenshot from a datasheet or a mockup of a datasheet if there are copyright concerns.) 

Worse of all, the text does not agree with the screenshot. (The screenshot shows the vcc pin as power input in the text above the screenshot it is passive.)

---

As this seems to be a symbol for a connector it can either be power input or output. 
It would depend on whether the connector does supply the rest of the schematic with power or if it connects to a separate system that gets supplied by the pcb currently being designed.

Passive would make sense if the designer of the symbol does not know what direction the power will flow.

Another thing: The VCC and GND pins should definitely have the same type!

ciampix, 2018-02-19:
In the meantime I fix the Italian version (no time for anything else) but I strongly suggest to fix the master (English) version...

poeschlr, 2018-02-19:
I opened a separate issue about this such that the problems discovered in the italian version can be handled separatly to the discussion about this tutorial in general.

ciampix, 2018-02-26:
Many thanks @poeschlr !