KiCad/kicad-doc #539: Confusing use of pin electrical type in "7.1 Using Component Library Editor"

poeschlr, 2018-02-19:
While discussing the italian translation of this section some issues in the english version came to light (original discussion: https://github.com/KiCad/kicad-doc/issues/537)

To save everyone a trip over to the other issue, here again my comments about it (A bit better structured):

---

Currently the tutorial states (with a lot of confidence) that some pins should have a specific electrical type. (Namely power in for VCC and passive for GND)
No reasoning is given why these types are selected. It might even be out of the intended scope of this part of the tutorial to go into this much detail.

---

My two suggestions:
State the scope of the tutorial and use passive for all pins. (Maybe remove the note about selecting specific electrical types from the text. Change the screenshot to show passive as this is the kicad default.) But also add a link to the part of the docu that explains electrical types.

If you want to explicitly show the usage of specific electrical types, then i would suggest to first introduce the intended component. An example would be to use a mockup datasheet to explain the requirements and give a reasoning behind the electrical type selection.

As this seems to be a symbol for a connector the VCC and GND pins can either be power input or output. 
Dependent on power flow direction. (Power out if the subsystem gets supplied from the currently designed pcb. Power in otherwise.)
Passive would make sense if the designer of the symbol does not know what direction the power will flow.

ciampix, 2018-02-27:
I totally agree with adding a link pointing to the part of the doc that explains the electrical types more in deep and convert/simplify the tutorial and related screen-shots in all passive pins. I would (as you stated) add a: "As symbol for a connector, the VCC and GND pins can either be power input or output, 
dependent on power flow direction. As we do not know it already (and besides it is outside the scope of this simple tutorial) let put them passive for all pins. For more information on the subject, see section blablabla..."

EeliK, 2019-01-12:
I think I can change it while I'm going through the Getting Started document. I don't find a way to assign this task to me.