KiCad/kicad-doc #559: Missing precompiled release packages post 4.0.0

zpuskas, 2018-04-15:
I'm maintaining kicad-doc package for Gentoo, however currently users can only compile the docs for themselves, which requires a latex installation. I'd like to add an option to download precompiled files, however the KiCad/kicad-doc repository is missing those release tar.gz files after version 4.0.0.

The `kicad-doc-HEAD.tar.gz` is not suitable for distro packaging (ever changing content and thus changing size, and hash). The rest of the 4.0.X releases at http://docs.kicad-pcb.org/ are individual files, which make it also make it unsuitable for packaging up.

Please provide official precomiled `kicad-doc-4.0.X.tar.gz` files (either Github or http://docs.kicad-pcb.org/ is fine). Though technically I could `wget` the the whole directory and host the tar.gz somewhere (even here on gihhub) but in order to keep authenticity and user trust providing them in the official repo would be the best.

jonasstein, 2018-04-16:
Would you prefer `http://docs.kicad-pcb.org/stable -> kicad-doc-4.0.X_all.tar.gz `, or
`http://docs.kicad-pcb.org/stable/en -> kicad-doc-4.0.X_en.tar.gz ` and so on for each LANG, or both?
Many distributions could use these tar balls.

zpuskas, 2018-04-17:
I've checked and most distributions do a per language package, so it would make the most sense to do that. For the Gentoo having per language packages would cause only a insignificant inconvenience (probably once, on initial ebuild creation), thus in the name of the greater good I'd say `kicad-doc-4.0.X_en.tar.gz` (and so on for each LANG and release) would be better.

nickoe, 2018-04-17:
Don't they download it once and package all package it from there?

zpuskas, 2018-04-17:
They probably do, however one has to crawl the entire directory tree. As for Gentoo, we always download from upstream and we'd prefer having a trusted upstream provided tarball. Thus having a `kicad-doc-4.0.X_en.tar.gz` or at least a combined kicad-doc-4.0.X.tar.gz` like what v4.0.0 has would make life easier for everyone.

nickoe, 2018-07-05:
They are now available on http://downloads.kicad-pcb.org/docs/

closed 2018-07-05 by nickoe