KiCad/kicad-doc #578: Getting Started: the MCU symbol

EeliK, 2018-05-30:
The MCU symbol used in the Getting Started Guide has been changed in the new library. This renders the screenshots of eeschema schematic out of date. Also many texts like "left", "right" etc. should be changed because the pins are in different places.

I see two possibilities:

1. Change everything according to the new symbol.

2. Attach the old symbol to the document in some way so that the users can put in into their filesystem and add to sym-lib-table.

Another related but easier problem is that the library system has changed and you don't need to add the official libraries manuallly any more, they are available by default (isn't that the case?) I made a pull request which changes the relevant section about as little as I thought was possible, still giving relevant information to the user, so that the symbol found in the file system is added manually even though it's already availabe. Should it be done differently?

EeliK, 2018-12-03:
I need an authoritave opinion or consensus about the MCU symbol. I was going through the document but it stalled, partly because I don't want to continue before this has been decided.

poeschlr, 2018-12-03:
My vote would be to update the tutorial with the current symbols (and footprints).

ciampix, 2018-12-03:
1) given that docs are already tagged, I think we are free to correct and update, so please go on.
2) please provide _small_ patches, for every chunk of text, so to not break docs compilation.

EeliK, 2018-12-03:
Yes, but what to do with the MCU in Getting Started? There are 6 large pictures in the "Using Eeschema" section which would need updating and then all the text which includes "left", "right" etc. If those are updated the document keeps the old structure. Other options could be adding the symbol as a text file (which would at the same time introduce how to add library files) or making it the example of how to use the symbol editor.

In the end it looks like the option one would be the easiest, although it means taking many screenshots and replicating the whole schematic.

ciampix, 2018-12-03:
I vote for option 1...

EeliK, 2018-12-03:
Ok, 2 against 0, so that's how it'll be done.

closed 2018-12-09 by EeliK