KiCad/kicad-doc #588: Documentation fails to build

GyrosGeier, 2018-06-04:
Hi,

the [automated build](http://darine.hogyros.de:8080/job/any-kicad-doc-head/) has been failing for a few days now:

    00:08:19.888 asciidoc: WARNING: cvpcb.adoc: line 6: missing language conf file: lang-zh.conf
    00:08:19.891 asciidoc: WARNING: cvpcb.adoc: line 6: missing section: [footer-text]
    00:08:27.250 a2x: ERROR: "dblatex" -t pdf -p "/etc/asciidoc/dblatex/asciidoc-dblatex.xsl" -s "/etc/asciidoc/dblatex/asciidoc-dblatex.sty" -P latex.output.revhistory=0 -P doc.publisher.show=0 -s /var/lib/jenkins/workspace/any-kicad-doc-head/CMakeSupport/pdf-cover-dblatex.sty -o /var/lib/jenkins/workspace/any-kicad-doc-head/Build/src/cvpcb/zh/cvpcb.pdf  "/var/lib/jenkins/workspace/any-kicad-doc-head/Build/src/cvpcb/zh/cvpcb.xml" returned non-zero exit status 1
    00:08:27.256 src/cvpcb/CMakeFiles/cvpcb_pdf_zh.dir/build.make:51: recipe for target 'src/cvpcb/zh/cvpcb.pdf' failed
    00:08:27.256 make[2]: *** [src/cvpcb/zh/cvpcb.pdf] Error 1

ciampix, 2018-06-04:
Sorry, it should build now.

ciampix, 2018-06-05:
There is something wrong in this automated build....
See the logs...
[...]
11:25:02 [ 98%] Built target plugins
11:25:02 Scanning dependencies of target pl_editor
11:25:02 [ 98%] Built target pl_editor
11:25:03 [ 99%] Built target gui_translation_howto_epub_ja
11:25:03 [ 99%] Built target gui_translation_howto_epub_de
11:25:03 Scanning dependencies of target gui_translation_howto
11:25:03 [ 99%] Built target gui_translation_howto
here hangs forever...

ciampix, 2018-06-05:
It compiles just fine in my pc...

ciampix, 2018-06-05:
Perhaps it is out of memory or disk space?

GyrosGeier, 2018-06-05:
No, the VM had lots of free memory, and there was a LaTeX process running at 100%.

GyrosGeier, 2018-06-05:
Hm, this seems to be reproducible.

GyrosGeier, 2018-06-05:
I have lots of errors compiling the fonts:

    kpathsea: Running mktextfm FandolSong-Regular/OT
    /usr/share/texlive/texmf-dist/web2c/mktexnam: Could not map source abbreviation O for OT.
    /usr/share/texlive/texmf-dist/web2c/mktexnam: Need to update /usr/share/texlive/texmf-dist/fonts/map/fontname/special.map?
    mktextfm: Running mf-nowin -progname=mf \mode:=ljfour; mag:=1; nonstopmode; input OT

(i.e. it's trying to build a font called "OT" instead of the proper font name)

At some later point, xelatex goes into an endless loop in line 590:

    ! Interruption.
    \@cline ...rulewidth \hfill }\cr \noalign {\vskip 
                                                      -\arrayrulewidth }
    l.590 \cline{1-1}
                     \cline{2-2}\cline{3-3}\cline{4-4}\cline{5-5}\endhead

That might be caused by missing font metrics.

System is Debian stable with "texlive-full" installed.

ciampix, 2018-06-06:
are you sure to have installed fonts-freefont-ttf?

ciampix, 2018-06-07:
Weird... I've set up a Debian Stable on Virtualbox just to be sure... it compiles everything like a charm...

GyrosGeier, 2018-06-07:
Okay, I'm stupid. That job still used Debian jessie (oldstable). Switched over to stable, changed a few dependencies around, should work now.

closed 2018-06-07 by GyrosGeier

ciampix, 2018-06-07:
ahhhhhhhhhh! :-)