KiCad/kicad-doc #603: Upgrading v4 to v5 document fails to compile, at least for Russian

GyrosGeier, 2018-06-25:
Hi,

from the autobuilder:

    15:26:23 [ 28%] Generating ru/kicad_upgrading_from_v4_to_v5.adoc
    15:26:23 po4a::asciidoc: Unable to parse attribute list: []
    15:26:23 src/kicad/CMakeFiles/kicad_translate_ru.dir/build.make:69: recipe for target 'src/kicad/ru/kicad_upgrading_from_v4_to_v5.adoc' failed
    15:26:23 make[2]: *** [src/kicad/ru/kicad_upgrading_from_v4_to_v5.adoc] Error 255
    15:26:23 CMakeFiles/Makefile2:9649: recipe for target 'src/kicad/CMakeFiles/kicad_translate_ru.dir/all' failed
    15:26:23 make[1]: *** [src/kicad/CMakeFiles/kicad_translate_ru.dir/all] Error 2

This is the first language that is attempted for this document, and it seems to happen while the translation is applied.

stambaughw, 2018-06-25:
Simon,

Is this the only language that is failing?  I'm not sure what even needs
to be corrected to fix the build error.  The error message isn't telling
me much.

Wayne

nickoe, 2018-06-25:
Could it be the ifdef::env-github[] stuff that it chokes on? What is that even? I think it is only for asciidoctor, we are still using plain asciidoc to compile the docs.

I will try to have a look when I get home.

stambaughw, 2018-06-25:
I'm not sure.  It was added by hauptmech.  One thing I did notice is
that I could see the warning and note icons in the github preview mode
which is kind of nice.  Without the env-github stuff, the github preview
doesn't show the icons.

ciampix, 2018-06-25:
For what I see on my PC, po4a does not behave nicely with macros like this one. For now I suggest to delete this piece of code as is github related. I'll post PR to the po4a dev list to add support for this piece of code (as I already have done with the poor table support). Unfortunately po4a support for asciidoc is not complete. It will be better in the future though.