KiCad/kicad-doc #616: README.md does not mention support for "LANGUAGES" cmake option

StefanBruens, 2018-07-25:
PR #567 added support for multiple languages, but README.md still only mentions the SINGLE_LANGUAGE option.

Updating this is trivial, but there is one caveat:
When using SINGLE_LANGUAGE, the package name is altered to include the language. So, is this really used? Should this be the concatenated language list, e.g. "-ca-de-en"?

nickoe, 2018-08-01:
I am not aware of anyone using the packages generated with the single language option. Maybe we should just indicate in the readme that it is deprecated i favour of the LANGUAGES variable.

StefanBruens, 2018-08-09:
@nickoe  updated PR to mention SINGLE_LANGUAGES is deprecated, and drop the special handling of the tar.gz package name.

nickoe, 2018-10-11:
Hmm, I guess this is closed by the above mentioend pull request.

closed 2018-10-11 by nickoe