KiCad/kicad-doc #618: KiCad documentation re-write?

evanshultz, 2018-07-26:
@ciampix 
WIth v5 just being released, there will be down time coming in the future before the next major version is closing in on release and new documentation is needed. Perhaps this gives an opportunity?

There is more polish that could be applied everywhere, and I proposed a whole new outline to the documentation some time ago. Would there be a time to revisit the entire documentation, deciding what to keep, what to move, what to re-write, what to polish, what to add, etc.?

I'd be interested in helping to take a fresh look at the documentation and working towards updating it for any future release target. I can dig up older posts I've made on the subject and also help with new ideas. If there is support for this, keep tag me so I can help. Thanks!

stambaughw, 2018-07-28:
While I agree that our documentation needs a lot of work, I would prefer
an incremental approach.  I just makes life easier for repo maintainers.
 There are parts of KiCad that are well documented and up to date.
Other parts, not so much so that is the best place to start.

Wayne

ciampix, 2019-01-14:
Since I think this is too generic as an enhancement wish... if there are no objections I will close this in a few days ... or just now... @stambaughw ?

nickoe, 2019-01-14:
I think it would be good to pick up evanshultz' suggestions. But yes, I would like if change are made in smaller hunks if at all possible. There are also many issues about new features that are added. I am not sure if it makes sense to try to document them first, or restructure and then review the new feature issues.

I guess it is also time to get rid of the cvpcb doc and merge it with pcbnew or eeschema where relevant.

I think the changes can be made on master as of now. If we decided that the new documentaiton is not well for 5.1, we can always branch out before as a post decision, although I expect the new doc if done soon will be compatible with 5.1, as that is essentially master of the kicad source now.

I am not sure why https://github.com/KiCad/kicad-doc/issues/394 restructuring died. I guess it is better we do something, rather than nothing at this point.

nickoe, 2019-01-14:
On a side note some work has also been done to integrate the docs more into the website. A preview is avalable, on http://docs.test.kicad-pcb.org/. This is still a work in progress, and if or when we swtich to asciidoctor I am not sure if that needs to be updated or rewritten. But essentiall it is generated by https://github.com/marekr/kicad-doc-website

evanshultz, 2019-01-14:
I'm happy to continue #394 or some other process. And you can see, my suggestion was to align the workflow with the tool and user needs, retaining existing section of the documentation where possible. Is there a better way to go about it?

I'm happy to contribute and think this is a fertile area for improving KiCad, but I would like broad acceptance on the direction so there's little time wasted and multiple people, if they're inclined, are able to join in.

nickoe, 2019-01-14:
For pull requests it is way easier to review and comment if one tries to make the changes atomic. What I mean by this is; if a section is to be moved and some wording updated. Then bulk mode the section, then commit with "Move foo to blah", then reword the change and add "Add explanation of dingleribs", and maybe a new commit later on that fixes various typos, this commit should also only be about fixing typos only, i.e. not updating some explanation.

If it makes sense, a auto reformat with line breaks should also be a separate commit. This will speed up reviews a lot. I when actually doing the writing it is also easier to review one's own change.

Keep in mind that this is just how I would like to see it happen. Don't consider this mandatory, but it really helps getting a response if changes are easily digestible.

ciampix, 2019-01-18:
For what is worth ... I totally agree with you @evanshultz and @nickoe

alfrede, 2019-03-03:
Hi all, i just began to help translating the Kicad Project. After a long night i came to this issue and have a question. All the translations which where done are going to the master branch? right? 
Why don't we use the change with Kicad 6 that we make an Standard for the Documentation. My Idea is that we make a branch after source ( https://github.com/KiCad/kicad-doc/tree/master/src) so that all the documents which are written till now goes to a Version 5 Branch (which actually doesn't exist) and all new to a saperatly new branch for 6 and all docs which where written in the time goes to the master.  I hope you understand what i mean.

nickoe, 2019-03-03:
Yes, active translations are on kicad-doc master. This is for master of kicad, what is currently for 5.1.0 of kicad. We only branch the docs after the fact, if it makes sense.