KiCad/kicad-doc #627: Help Manual opens in editor - KompoZer

nilands55s, 2018-09-21:
This may not be a problem with KiCad or even the Documents, but I need a place to start. When using the help menu to open the Eeschema or KiCad Manual they open in the editor KompoZer that had been installed long before either Eeschema or KiCad. In example /usr/share/doc/kicad/help/en/eeschema.html will open in the KompoZer HTML editor. The Properties of eeschema.html are set to open FireFox web browser according to nautilus file manager. Doing a google search on the problem only results in how to install KompoZer a program left over from Ubuntu 12.04 LTS but I keep in 14.04.5 as it is easy to use. I don't believe  KompoZer is any longer supported, so I need to figure out how to troubleshoot this problem. If someone could get me started this would be great!
Thanks James

nickoe, 2018-09-23:
Mm,  what happen if you run `xdg-open /usr/share/doc/kicad/help/en/eeschema.html', does it open in forefox or kompozer?

ciampix, 2018-09-27:
That seems to me a Linux/Ubuntu problem anyway. I vote for close this if not proven the contrary...

nickoe, 2018-10-09:
No response.

closed 2018-10-09 by nickoe