KiCad/kicad-doc #629: CvPcb Manual

LarryJoy, 2018-10-19:
Comments as follows:
5.2.5. Environment Variable Substitution
1st bullet point, "• The KIPRJMOD environment variable. This points always the current project..." I think this would be better as, "...This always points to the current project..."

5.2.6. Using the GitHub Plugin
1st paragraph, last line, "...in the footprint library table row a must be set..." Delete the "a".

5.2.7. Usage Patterns
Last (8th) paragraph, 2nd line, "...only require for the project..." Think "require" should be "required".

7.2 Filtering the Footprint List
1st paragraph, 1st line, "...is highlighted when the one or more of the filtering option is enabled..." Think this should be, "...is highlighted when one or more of the filtering options is enabled..."

ciampix, 2018-10-21:
Good catch! I'll fix it in a couple of days, thanks!

ciampix, 2018-10-22:
Committed, thanks!

commit b8e5f446 (2018-10-22): Fixed some spelling errors, thanks to LarryJoy for the revision, fixes #629

Fixed some spelling errors in the original (English) text and de-fuzzified
the corrisponding strings in the other language files.
closed 2018-10-22 by ciampix