KiCad/kicad-doc #630: Pcbnew feedback/comments

LarryJoy, 2018-10-25:
2.3.3. Initial Configuration
Paragraph after the 1st Tip, "The **first **thing****to when configuring KiCad do is to..." Should be, "The **first **thing****to do when configuring KiCad is to..."

6.1.3. Reading the netlist generated from the schematic
Note, 2nd line, "...changing a 1/8W resistance to 1/2W..." Should be, "...changing a 1/8 W resistor to 1/2 W..."

8.5.2. Standaard
1st bullet, "• Clearance: 0.35mm..." Should be "• Clearance: 0.35 mm"
2nd bullet, "• Track width: 0.5mm..." Should be "• Track width: 0.5 mm..."

10.3.2 Clearance and minimum copper thickness
Paragraph, 2nd line, at end, "...copper ares." Should be, "...copper areas."

10.6 Editing zone parameters
Last sentence, "If the zone is already filled, refilling it will be necessary." ? What is meant? Should this be, "If the zone is already filled, refilling will not be necessary." ? Or, "If the zone is not filled, refilling will be necessary." ? Please clarify.

10.8 Change zones net names
2nd sentence, "When a global DRC control is made..." ? Delete the word "control".

10.10 Creating a Keepout area
Last paragraph, "...with no copper pour will be not filled." I think this would be better as, "...with no copper pour will not be filled."

evanshultz, 2018-10-26:
@LarryJoy 
Would you mind to create a PR for these changes? That would make it a bit faster to review and get the changes pushed to the documentation.

LarryJoy, 2018-10-26:
Evanshultz,I am signed up for Ubuntu One but I am confused as to how to do a PR (pull request). I did get an e-mail message from Marco Ciampa that said, "Thanks, as usual wait a couple of days for the application." So I think my comments are being reviewed already.
Regards, Larry Joy

evanshultz, 2018-10-26:
Yes, I read Marco's comment above but you have 5 closed issues already of the same format. If you will submit PRs it will be easier to merge your changes.

You don't need an Ubuntu One account to submit a PR. Googling "github submit pr" turns up a lot of results including nice step-by-step instructions from GitHub as the first result.. If you want to give it a shot many folks here will be able to help you with this process. Installing the GitHub Desktop app, if you run Windows or Mac, should be an easier way to get started than just the command line.

LarryJoy, 2018-10-29:
Evanshultz, For me there always seems to be a hiccup in trying to follow written instructions. I will need some additional help in knowing what inputs are needed at specified times. Here are the instructions I have printed off for "Creating the pull request":
1. On GitHub, navigate to the main page of the repository.
2. In the "Branch" menu, choose the branch that contains your commits.
3. To the right of the Branch menu, click New pull request.
4. Use the base branch dropdown menu to select the branch you'd like to merge your changes into, then use the compare branch drop-down menu to choose the topic branch you made your changes in.
5. Type a title and description for your pull request.
6. Click Create pull request.
1. I have a GitHub account with user ID and Password. On the main screen I go to Repositories > KiCad/kicad-doc. I think I am in the right place, if not let me know where and how to get to where I should be.
2. The box on the left has "Branch: master" with a filled triangle pointing down. I click on the down pointing triangle and get "Switch branches/tags". The choices below a blank box are "4.0", "dynamic_version_info", and "master". What should I choose or what should I type in the blank box? At this point I would say there is no edits>saved changes>commits, so I don't know what is going on here?
3. When I click on "New pull request" the screen now has "base: master" and "compare: master". Below this is "Create pull request" with a green background.
4. When clicking on the down triangle of "base: master" it now has "Choose a base branch". The choices are "4.0", "dynamic_version_info", and "master". What should I choose or what should I type in the blank box?  When I click on the down triangle of "compare: master" it has "Choose a head branch" with choices of "4.0", "dynamic_version_info", and "master". What should I choose or what should I type in the blank box?

That is as far as I can get, which isn't very far. I need more complete instructions. Help!

Regards, Larry Joy

evanshultz, 2018-10-30:
Hi Larry,

Perhaps it is best to write the corrections and suggestions here as you've done. I'd rather you feel you're able to submit your thoughts and they're accepted than to cause any frustration for you.

Let me know if you prefer to maintain status quo or not and we'll go from there.

LarryJoy, 2018-10-30:
evanshultz,
Yes please, the status quo as I have submitted these grammar and spelling errors edits. That is the way the instructions are in the "manual" when it is stated, "Please direct any bug reports, suggestions or new versions to here:
    About KiCad document: https://github.com/KiCad/kicad-doc/issues"
Regards, Larry Joy

ciampix, 2018-10-31:
Grammar and spelling errors in the reference language (i.e. English) are a bit different from writing a new chapter to be translated. They potentially create a huge amount of useless work. If one wants to correct some sentences they need also to provide the update and unfuzzy of all the current translations. If the PR is for the English only, I prefer to do it by myself by hand. If one wants to contribute with a patch and a pull request they have to include also the patches for all the different language .po files. Yes I know, I probably have to clarify these concepts the the translation guide...

nickoe, 2018-11-28:
Do we intend to get any of the feedback here included in the 5.0.2 docs? If not I would like to tag it very soon.

ciampix, 2018-11-29:
Yes sorry, my bad, I almost forgot it. Do I still have time to do it? If not, not a big deal, I'll do it for the next release...

nickoe, 2018-11-29:
Lets say we will tag it saturday. It looks like the libs are not tagged yet anyway.

ciampix, 2018-11-29:
Ok thanks!

ciampix, 2018-12-01:
Thanks Larry, I left only this:

10.6 Editing zone parameters
Last sentence, "If the zone is already filled, refilling it will be necessary." ? What is meant? Should this be, "If the zone is already filled, refilling will not be necessary." ? Or, "If the zone is not filled, refilling will be necessary." ? Please clarify.

is this an error?

commit 4d0097e7 (2018-12-01): Fixed some spelling errors, thanks to LarryJoy for the revision, fixes #630

Fixed some spelling errors in the original (English) text and de-fuzzified
the corrisponding strings in the other language files.
closed 2018-12-01 by ciampix