KiCad/kicad-doc #638: Asciidoctor? UI elements?

EeliK, 2018-12-14:
Asciidoctor apparently wasn't ready enough in 2015, but how about now? I found a couple of nice features there, although they are "experimental" (need :experimental: attribute).

Check out https://asciidoctor.org/docs/asciidoc-syntax-quick-reference/#ui-macros. It would be great to have UI elements in the text. Now we are using *bold* and  _italic_ for everything. Just think how much more clear and easy to read (and good looking) it was if it had buttons and keys which would be distinct from other text.

ciampix, 2018-12-15:
Yes this is very interesting, thanks for pointing it out. Migration from asciidoc to asciidoctor build was planned from the start on some time in the future. But first we have to create a working environment able to create documents output complete as it is now. I mean, pdf, html and epub with all features like cover, content indexes and so on. We can create a fork for the built environment and see how far we are able to go, why not?

ciampix, 2018-12-16:
And btw I spotted a very tricky bug of dblatex that happens in unspecific circumstances that makes me waste a lot of time (and makes me crazy) that gets triggered by some quoted text messages here and there... the only turnaround is to replace the symmetric quotes with asymmetric ones (like those replaced by the handy `` ... '' macros of asciidoc). So I am more than eager to replace asciidoc with asciidoctor in the build scripts... I'll manage to make a new build environment in the near future... I promise...

ciampix, 2019-01-14:
Why not start adding (marked as experimental) an option to the cmake configuration file to check for the presence and compile with asciidoctor? In this way we can start using it and prepare for the migration...

marekr, 2019-02-27:
Technically, the doc website is using asciidoctor on the already translated adocs. So it works fine fyi.