KiCad/kicad-doc #642: explanation on making new symbol i

atzed, 2018-12-26:
The helpfile for creating a new component is unclear. The used symbol for storing the new component does not exist (green Arrow down in opamp)
See getting started with kicad page 37/46 point 9 to 14. It just doesn't work.
The -- Preferences -> Component Libraries -- link does not exist either.
When I click -- Create New Symbol --- I have to choose an existing library where I think I was ment to make a new one.
Regards Atze.

poeschlr, 2018-12-26:
If this is a cry for help about getting started with making symbols than i can provide you with an alternative to the official library. The FAQ of the forum has some hopefully interesting entries about that:
- https://forum.kicad.info/t/tutorial-how-to-make-a-symbol/13336
- https://forum.kicad.info/t/creating-a-new-symbol-library/13341

atzed, 2018-12-27:
Thanks for the reply this is exactly what i was looking for but could not
find.