KiCad/kicad-doc #643: Offering Fix for Debian Build

codemonkeyrawks, 2019-01-03:
Attached is supporting documentation for installing KiCad on Debian. I had to rename the doc to get this to attach (lines: 123 - 138).

[Getting Started in KiCad](https://github.com/KiCad/kicad-doc/files/2722395/getting_started_in_kicad.txt)

codemonkeyrawks, 2019-01-03:
__________________________________________________

Under Debian, the easiest way to install backports build of KiCad is:
__________________________________________________

# Set up Debian Backports
echo -e "
# stretch-backports
deb http://ftp.us.debian.org/debian/ stretch-backports main contrib non-free 
deb-src http://ftp.us.debian.org/debian/ stretch-backports main contrib non-free
" | sudo tee -a /etc/apt/sources.list > /dev/null

# Run an Update: Install KiCad
sudo apt-get update
sudo apt-get install -t stretch-backports kicad
__________________________________________________

ciampix, 2019-01-03:
I'm seeing to it now, please be patience, thanks!

closed 2019-01-03 by ciampix

codemonkeyrawks, 2019-01-04:
Thanks @ciampix .

commit 4a653f9c (2019-01-03): Updated Debian install procedure in GSIK manual, fixes #643