KiCad/kicad-doc #645: Problem with identifiers containing "->" string

needfulthing, 2019-01-12:
I'm currently working on an enhanced translation of the German version of "getting_started_in_kicad" and found that all msgid sections containing "->" do not get translated in the finished docs. Example:

```
#. type: Plain text
#: getting_started_in_kicad.adoc:257
#, fuzzy
#| msgid ""
#| "To view hotkeys within any KiCad tool go to *_Preferences -> Hotkeys -> "
#| "List Current Keys_* or press the question mark (\"?\"):"
msgid ""
"To view hotkeys within any KiCad tool go to *_Help -> List Hotkeys_* or "
"press Ctrl+F1:"
msgstr ""
"Sie können eine Liste der möglichen Tastaturbefehle in jedem KiCad Werkzeug "
"einsehen, indem Sie *_Hilfe -> Tastaturbefehle auflisten_* aufrufen oder "
"einfach Strg+F1 auf der Tastatur eingeben."
```

These sections are still in English in the German version of the document. Probably this is a bug in the used documentation system. I'm only replacing the translated strings in the file de.po and have no further knowledge about AsciiDoc and what else in used to build the docs, but I assume, replacing "->" with something else might fix this error.

This error seems also to be triggered by other specific msgid strings. The following section also does not get translated (among a lot of others in the section 4.1. of the document):

```
#. type: Plain text
#: getting_started_in_kicad.adoc:326
#, fuzzy
#| msgid ""
#| "Click on the middle of your schematic sheet. A __Choose Component__ "
#| "window will appear on the screen.  We're going to place a resistor. "
#| "Search / filter on the 'R' of **R**esistor.  You may notice the 'device' "
#| "heading above the Resistor. This 'device' heading is the name of the "
#| "library where the component is located, which is quite a generic and "
#| "useful library."
msgid ""
"Click on the middle of your schematic sheet. A __Choose Symbol__ window will "
"appear on the screen.  We're going to place a resistor. Search / filter on "
"the 'R' of **R**esistor.  You may notice the 'Device' heading above the "
"Resistor. This 'Device' heading is the name of the library where the "
"component is located, which is quite a generic and useful library."
msgstr ""
"Klicken Sie in die Mitte vom Schaltplan. Ein __Bauteilauswahl__ Fenster wird "
"sich öffnen. Wir werden einen Resistor (Widerstand) hinzufügen. Im "
"Filterfeld geben Sie 'R' ein für **R**esistor. Sie werden den Titel 'device' "
"über dem Resistor bemerken. Dieser Titel 'device' ist der Name der "
"Bibliothek in dem das Bauteil zu finden ist, welche eine generische und "
"nützliche Bibliothek ist."
```

nickoe, 2019-01-13:
I don't think this has anything to do with the arrow thing, it is simply because the translation is marked as fuzzy. If you unfuzzy it, it should be good. Of course when you do that you need to check that the translated version still match the english.

needfulthing, 2019-01-14:
Aaaah, thank you so much,that totally makes sense :)

needfulthing, 2019-01-31:
Going forth in the new German translation I now found sections that do not contain the fuzzy mark, but still do not get translated, e.g.

```
#. type: Plain text
#: getting_started_in_kicad.adoc:533
msgid ""
"Let's now deal with unconnected wires. Any pin or wire that is not connected "
"will generate a warning when checked by KiCad. To avoid these warnings you "
"can either instruct the program that the unconnected wires are deliberate or "
"manually flag each unconnected wire or pin as unconnected."
msgstr ""
"Kommen wir nun zum Umgang mit nicht verbundenen Signalen. KiCad wird bei der "
"Schaltplan Überprüfung für jeden Pin oder Leitung/Signal welche(s) nicht "
"verbunden ist eine Fehlermeldung generieren. Um diese Fehlermeldungen zu "
"vermeiden müssen wir KiCad bewusst mitteilen das diese nicht verbundenen "
"Signale keine Fehler sind."
```

Maybe there is still another problem.

needfulthing, 2019-01-31:
Doh, please ignore this, I accidentally looked at the original version of the doc.