KiCad/kicad-doc #655: Class Designation Letters

LarryJoy, 2019-02-07:
Clause S6.1 - _Component Reference Designator (RefDes)_ of KiCad Library Convention (KCL). These are incorrectly called reference designators when, in fact, these are properly called Class Designation Letters or just Class Letters of a reference designator. As we refer to IPC standards for PCBs there are standards for assigning reference designators. From ANSI/ASME Y14.44, Clause 1.7 _Definitions_, a "basic reference designation: the simplest form of a reference designation consisting only of a class letter portion and a number (namely, without mention of the item within which the reference-designated item is located)." Can the nomenclature be changed in KiCad V6?

The official list of Class Letters is from IEEE 315 as follows:
[image]
In order to see the complete "official" list I refer you to my series of articles at <techexplorations.com/reference-designations-for-electrical-and-electronics-parts-and-equipment>. Two entries in the S6.1 table that need to change or be added are "M" for motor, M is the class letter for a meter while the class letter B is for a motor.
[image]
[image]
There are other anomalous entries as well and I again refer you to my series of articles that includes a list of anomalous class letters and what class letters should be used.
--Larry

evanshultz, 2019-02-07:
Larry,

I don't intend for this to be too much or aggressive, but I do want to be comprehensive. Also, I'm not the head librarian but I'm compelled to chip in early to (hopefully) steer any potential conversation appropriately.

There are a number of documents that state ref des prefix for various part types, but taken as a whole this part of the industry is a mess with little overlap between competing documents. There is undoubtedly more conflict than agreement when comparing "standards" for ref des prefix.

The official KiCad library is designed to suit many users well but is not specific to any one PCB manufacturing/assembly process or one country/geographical location, two give two examples. The current ref des prefix scheme has been built up by the librarians and contributors over the years from all over the world with varying experiences and, in the years I've been on the team, I've seen few objections. My experience is that this is common and one should expect variation in ref des prefix. I've never worked for a company, or heard second hand of a company, that stayed true to one ref des prefix scheme. Colloquially, this indicates things are generally OK as they are.

Let's look at switching to using the term "Class Designation Letters" or "Class Letters". Leaving out the word "reference designator" or "ref des" means it's less obvious what is being discussed. And with KLC and KiCad we also need to consider clarity for many people who are not native English speakers. I believe "ref des prefix" is more common and understandable. To the mention the KiCad interface uses the word "Reference".

You brought this general topic up several times in the library repo. Let's use https://github.com/KiCad/kicad-symbols/issues/808 as an example. You brought up a 1-3 letter system which @poeschlr point out is uncommon and would be far more awkward for KiCad users that the status quo. I liked your stop sign analogy, but the conversation ended with the head librarian stating that KiCad's official library will go it's own way as the current scheme falls within industry norms. And it's clear to me, both at the time and when re-reading it now, that following IEC strictly is not gonna happen.

Your post above shows me you are especially passionate on this topic. I hope you to continue to contribute to the documentation, KLC, and library as you are able and willing. KLC should be flexible and change with the times and evolution of the industry so this topic, and others, can and should be revisited in the future. But for now, strict adherence to the IEC ref des system has was not accepted by the librarian team and I am still against it.

With regards to your specific example, I would not like to change the motor ref des prefix to "B". Using "M" is almost surely more clear and obvious for KiCad users. And as just stated, the library has not taken the position to strictly follow IEC 315 (or any other "standard") for ref des prefix.

Lastly, a couple of procedural notes:
1. KLC is contained on the website so PRs or issues about KLC can be directed at https://github.com/KiCad/kicad-website/tree/master/content/klc. However, for discussion like the above which isn't settled and is a request, https://github.com/KiCad/kicad-symbols is probably the best place to get visibility and connect with the right folks (the librarian team).
1. Updating KLC is separate from the KiCad releases. Sometimes we will target an implementation of a change to coincide with a release, but most of the time KLC can be updated once consensus is reached.

poeschlr, 2019-02-07:
Clarification on your last point (marked with 2): If a change to the KLC would require renaming of symbols than such a change can only happen with a major kicad release.

---

Regarding refdes in general:
We use the name reference to describe this in the KLC because kicad uses this name in the graphical interface. It would be confusing to kicad users if the KLC uses a different terminology when compared to the software interface. If you want the software interface to change then please open a bug report over at launchpad.

---

Regarding standardization:
My experience agrees with the one by evan. Especially with regards to motors. I have never seen the letter B used for motors. (I have seen that letter used for sensors. One could argue that sensors are kind of the opposite of motors.) In every schematic i ever read motors where given the letter M. (Makes sense to be honest. After all the word "motor" starts with that letter.)

The fact that the standard is not really followed by anyone might be a pointer to the standard being made badly. (Example: Where is the connection between B and motors?) Another reason might be that the ref des seems to be mostly used to group similar components in the BOM not to convey information (neither to the reader and most definitely not to machines.) The information about a parts function is for the most part communicated by the form of the symbol. Or for more complex parts by use of pin names.

LarryJoy, 2019-02-08:
[DEEP BREATH] [Begin Rant]
Yes, I am passionate about the subject matter. It is a matter of knowledge gained through education and training. I think there is some confusion on your part as you reference IEC 315, which is actually IEEE 315. In the U.S. the American National Standards Institute (ANSI) is made up of standards setting bodies such as the Institute of Electrical and Electronics Engineers (IEEE), the American Society of Mechanical Engineers (ASME), the American Welding Society (AWS), the Association Connecting Electronics Industries (previously the Institute for Printed Circuits) (IPC), etc. ANSI does not write standards themselves but has a process whereby the standards setting body gains approval to be accepted as an "ANSI" standard. Most standards are accepted as norms and some standards are mandatory as they are written into the laws governing the subject matter. One such area is elevators for public use. Another area is EMI and EMC standards. In order to keep a particular standard in an approved ANSI status the standard must be looked at every five years and reaffirmed (R) as is or changed or revised as necessary.

Now to the particular ANSI approved standards covering reference designation methods and the class letters used in the formation thereof. IEEE 200 _IEEE Standard Reference Designations for Electrical and Electronics Parts and Equipment_ started out as ANSI Y32.16-1968, was board approved 1975-10-31, published date of 1975-12-30, reaffirmed date of 1988-10-20, and withdrawn 1997-01-17. Appendix B made mention of Item Designations per IEC Publication 113-2. I don't know why the IEEE dropped support of this standard except to say that they had bigger fish to fry. ANSI approved ASME Y14.44-2008 _Reference Designations for Electrical and Electronics Parts and Equipment_, reaffirmed date of 2014, took the place of IEEE 200. The ASME actually paid the IEEE and bought the standard. The Unit Numbering Method and its two derivatives have thus been around for 50 years.

From what I can find out IEEE 315 _Graphic Symbols for Electrical and Electronics Diagrams (Including Reference Designation Letters)_ started out in 1971, board approved 1975-09-04, published 1975-12-01, and reaffirmed 1993-12-02. Although the IEEE still considers this standard as active, they keep the copyright on it and offer it for sale, the standard is no longer ANSI approved as there have been no reaffirmations, revisions, or updates to it for many years, probably a couple of decades. The piece of this standard that goes with the reference designation method is Clause 22 _Class Designation Letters_. This standard has been around for 47 years. IEEE 315A-1986 is a supplement to and is included with IEEE 315. The supplement added graphic symbols from what is now IEC 60617 and may be the reason the IEEE does not do anything to revise or update IEEE 315.

What I am trying to say with all of this history information is that the standard(s) for applying reference designations have existed, been around for a long time, have been updated when necessary, and they work. The problem is the information has been lost. Teachers, instructors, and professors don't teach it because they don't know about it or deem it not that important. It seems as if it should be tribal knowledge.

Remember the old squirrel cage blower fans? Because the class letter M was assigned to mean a meter the class letter B was used for a "blower" and it had a motor, thus class letter B is for a motor. Why do we use L for an inductor? Because C is for a capacitor and not a coil and I is for current. The class letter to use for "sensor (transducer to electric power)" is A. In ASME Y14.44, Clause **1.4 Changes in Reference Designation Class Letters** has "If changes in the state-of-the-art appear to make changed or new reference designation class letters necessary, provide the following information:
(a) item for which new designation is requested, type or class of item, and frequency of use (per drawing, per set, etc.)
(b) present reference designation used.
(c) proposed reference designation
(d) reasons for recommending a change (supply source material, if feasible)
(e) graphic symbol
Address comments to The American Society of Mechanical Engineers, Secretary, Y14 Standards Committee, Three Park Ave., New York, NY 10016, and refer to this Standard."
So if someone in the U.S. wanted to add FB (for ferrite bead) to the official class letter list they could follow the above procedure. After doing all this typing I guess the question is, why doesn't the committee have their ear to the ground, be pro active, and initiate changes themselves? I'll have to look into this.

I recently did a project where I did a top level schematic diagram that had three assemblies that made up the the single unit set. The unit had a Front Panel Assembly, a Rear Panel Assembly, and a printed board assembly (PBA). I assigned reference designator "prefixes" as follows:
A1 for the Front Panel ASSY
A2 for the Rear Panel ASSY
A3 for the PBA
In other words there was an A1R1, an A2R1, and an A3R1 and the ref des prefix let you know which assembly the part was used on. The associated parts list (PL) was all in alpha numeric order, of course. In order to get that portion of the schematic diagram for the PBA board layout, I copied the schematic over to another file (.pro) project, deleted all parts that had the A1 and A2 ref des prefixes and then deleted the A3 prefix so I was left with basic reference designators. I was then able to layout a PCB.

If you liked the stop sign story here is another one. A driver is driving out in the country when he comes to the outskirts of a town, wherein he reads a sign that says "In this town red means go and green means stop". Do you think the driver can get through the town without an accident? After all he has all the information he needs presented at the town boundary.
[End Rant]
--Larry

closed 2019-02-08 by LarryJoy