KiCad/kicad-doc #670: dblatex-pdf-zh.xsl

taotieren, 2019-02-25:
`Scanning dependencies of target idf_exporter_pdf_zh
[ 74%] Generating zh/idf_exporter.pdf
[ 74%] Built target getting_started_in_kicad_epub_pl
Scanning dependencies of target idf_exporter_pdf_ru
[ 74%] Generating ru/idf_exporter.pdf
a2x: ERROR: "dblatex" -t pdf -p "/etc/asciidoc/dblatex/asciidoc-dblatex.xsl" -s "/etc/asciidoc/dblatex/asciidoc-dblatex.sty" -P latex.output.revhistory=0 -P doc.publisher.show=0 -s /home/travis/build/KiCad/kicad-doc/CMakeSupport/pdf-cover-dblatex.sty -b xetex -p /home/travis/build/KiCad/kicad-doc/src/idf_exporter/../../xsl/dblatex-pdf-zh.xsl -o /home/travis/build/KiCad/kicad-doc/build/src/idf_exporter/zh/idf_exporter.pdf  "/home/travis/build/KiCad/kicad-doc/build/src/idf_exporter/zh/idf_exporter.xml" returned non-zero exit status 1
make[2]: *** [src/idf_exporter/zh/idf_exporter.pdf] Error 1
make[1]: *** [src/idf_exporter/CMakeFiles/idf_exporter_pdf_zh.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 74%] Built target idf_exporter_pdf_en
[ 74%] Built target idf_exporter_pdf_pl
[ 74%] Built target idf_exporter_pdf_ru
make: *** [all] Error 2
The command "mkdir build && cd build && cmake .. && make -j4" exited with 2.`
Only compiling idf_exporter_pdf_zh will cause problems, compiling other  *_zh.*  will not cause problems.

taotieren, 2019-02-26:
[https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=622359;msg=15](url)

tijuca, 2019-02-27:
I can't reproduce this with current HEAD.

nickoe, 2019-02-27:
@taotieren Is this issue resolved?

ciampix, 2019-02-28:
I think this is resolved or missing data, I'm closing it.

closed 2019-02-28 by ciampix