KiCad/kicad-doc #681: Getting Started Doc: section 4.1, step 58

finson, 2019-03-04:
The doc says:

> KiCad automatically generates the command, for example:
xsltproc -o "%O" "/home/<user>/kicad/eeschema/plugins/bom2csv.xsl" "%I"
You may want to add the extension, so change this command line to:
xsltproc -o "%O.csv" "/home/<user>/kicad/eeschema/plugins/bom2csv.xsl" "%I"

At least in my case, when I did not add the extension, the BOM process failed:
> Run command:
xsltproc -o "D:/users/finson/repos/kicad/projects/jumperoo/stem-module" "C:\Program Files\KiCad\bin\scripting\plugins\bom2csv.xsl" "D:\users\finson\repos\kicad\projects\jumperoo\stem-module.xml"

>Command error. Return code 11
Error messages:
I/O error : Permission denied

Adding the .csv extension fixed the issue.

> Run command:
xsltproc -o "D:/users/finson/repos/kicad/projects/jumperoo/stem-module.csv" "C:\Program Files\KiCad\bin\scripting\plugins\bom2csv.xsl" "D:\users\finson\repos\kicad\projects\jumperoo\stem-module.xml"

>Success

I'm not sure what exactly is failing (maybe a file rename?) but if this is true on all or many other systems, the doc at least should be updated to be more definite about adding the extension and why.  The best fix would be in the code so that the default command doesn't generate an error, of course.

Doug Johnson

--------------------------------

My system:  

OS:   Windows 10 Pro, v1803, OS build 17134.590

Application: kicad
Version: (5.0.2)-1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.61.1 OpenSSL/1.1.1 (WinSSL) zlib/1.2.11 brotli/1.0.6 libidn2/2.0.5 libpsl/0.20.2 (+libidn2/2.0.5) nghttp2/1.34.0
Platform: Windows 8 (build 9200), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8)
    Boost: 1.68.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.61.1
    Compiler: GCC 8.2.0 with C++ ABI 1013

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=OFF
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

evanshultz, 2019-03-04:
@finson 
Thanks for the report.

I believe this dialog and its functionality has changed dramatically in v5.1, as it looks like this is in a nightly from late last week:
[image]

For me, a BOM is created without issue and the extension is present by default.

5.1 should be tagged very soon and hopefully announced in a week or so. Once that's done, updating the documentation to reflect the current KiCad would be a welcome contribution.

finson, 2019-03-04:
Thanks Evan.  Will be mindful of the next release and check the doc to make an update.

Doug

[image] [https://user-images.githubusercontent.com/1936989/53757933-c3e06380-3e71-11e9-8922-5877a1c3f7fd.png]