KiCad/kicad-doc #685: Use the same logo as website on PDF?

marekr, 2019-03-06:
So right now the generated PDF has a goldish form of the logo in a different font.

Any reason it's not the "main" blue and white logo?

stambaughw, 2019-03-06:
We should be using the new KiCad logo image everywhere in place of the old logo.

ciampix, 2019-03-06:
Sorry I'm a bit "new"... which one is the "new logo"?

stambaughw, 2019-03-06:
@ciampix the blue logo.

ciampix, 2019-03-07:
Ok so it is wrong even in the program icon and the about window! I suggest to fix it in the next upcoming 6 version though... it's too late for this change in the 5.2 I am afraid...

alfrede, 2019-03-07:
After all my mistakes i do, i think for version 6 we should rewrite the "README for doc authors".

I will write down my experiences.

ciampix, 2019-03-07:
@alfrede yours is a very good idea™

ciampix, 2019-03-07:
Even in the bug tracker the icon is the old one...

nickoe, 2019-03-07:
@stambaughw I don't have access to edit it on the launchpad page, but I have attached a 400px version for your convince.

I suggest to use the Ki for the square icons.
[image]

nickoe, 2019-03-07:
@marekr I guess the main reason is that there was a time someone wrote some docs, this in a time where people were inspired by libreoffice designs... and then no one dared to change it. I think the blue KiCad logo is widely used and recognized, so we should of course use it for the title page of the docs. I have had the intention to change it for a long time, but never got around to it.

marekr, 2019-03-08:
@nickoe Yea that's fine. I was just bringing it up. It looks like it should be changed for 5.1 just fine? It doesn't affect translations and it's just a logo.png in each doc folder.

nickoe, 2019-03-08:
Yeah, xeel free to pr it

nickoe, 2019-03-08:
Woopsie, fixed in 5ed4c47

closed 2019-03-08 by nickoe