KiCad/kicad-doc #696: Building issue on Ubuntu 18.10

jsreynaud, 2019-03-11:
On Ubuntu 18.10 and 19.04 there is a building issue for pcbnew_pdf_ja target.
When removing extra quote on src/pcbnew/po/ja.po:2836 it fix the compilation issue:
=============================================================
diff --git a/kicad-doc/src/pcbnew/po/ja.po b/kicad-doc/src/pcbnew/po/ja.po
index 821076f3..d27d80d4 100644
--- a/kicad-doc/src/pcbnew/po/ja.po
+++ b/kicad-doc/src/pcbnew/po/ja.po
@@ -2836,7 +2836,7 @@ msgid ""
 "Down` and `Space bar`). The `Escape` key generally cancels an operation in progress."
 msgstr ""
 "キーボード（ファンクションキー `F1`, `F2`, `F3`, `F4`, `Shift`, `Delete`, `+`, `-`, `Page Up`, "
-"`Page Down` 及び `スペース バー`）。`エスケープ` キーは、一般的に進行中の操作を取り消します。"
+"`Page Down` 及び スペース バー）。`エスケープ` キーは、一般的に進行中の操作を取り消します。"

 #. type: Plain text
 #: pcbnew_general_operations.adoc:27
=============================================================
Anyway I don't speak/read/understand japanese... So not sure it's a good solution.

nickoe, 2019-03-11:
@starfort-jp Are you able to comment on this?

starfort-jp, 2019-03-12:
Hi all,

In Japanese, this kind of quote has little meaning.(usually use brackets instead of quote)  But if the translator see this sentence, he/she might be misunderstood it's a mistake in translation.(Because the original text includes "quote.")

I think one solution is removing all quote on this sentence - like this.
[image]
If it fix the compilation issue, I think it should be ok.

tijuca, 2019-03-18:
https://lists.launchpad.net/kicad-doc-devs/msg00128.html and ff
It's a problem within the asciidoc source.

I've "fixed" this by a small patch to the Japanese po file for Pcbnew that simply removes the translation so the users will see here the original English part.

ciampix, 2019-03-19:
I think that it depends on the type of quote used for the translation. Based on this it should not be so difficult to test this theory...

tijuca, 2019-03-19:
https://lists.launchpad.net/kicad-doc-devs/msg00154.html
```

The reason is the recompile regexp within asciidoc is a bit to stupid to
detect the backticks signs are needed in the original string and is
breaking the original string into multiple parts which fails later due
missing controlling escaping character.
```

ciampix, 2019-03-19:
Ok so it should be easy to fix just adding spaces into the right places without getting rid of the whole translated string or not?

starfort-jp, 2019-03-19:
Thanks for the explanation. I understand that "`"(backquote/backtick/grave accent) behaves like "\"(escape character) in C.  I've added spaces and created a pull request. Could you please confirm it?

ciampix, 2019-03-19:
Confirmed and committed, thanks!

closed 2019-03-19 by ciampix

starfort-jp, 2019-03-21:
Thank you for confirming.