KiCad/kicad-doc #697: po4a chokes on images + text in tables

StefanBruens, 2019-03-12:
Po4a has isses when processing the "eeschema_simulator.adoc", the table with the "image::images/opamp_symbol.png"

The following minimal example shows the issue:

```
|====
|First col
|Second col

image::foo.png[alt="Alt"]

Text below image 1
|====

image::foo.png[alt="Alt"]

Text below image 2
```
After processing with po4a, e.g. `po4a-normalize -f asciidoc test.adoc`, this ends up as

```
|====
|First col
|Second col

image::foo.png[alt="Alt"]
  Text below image 1
|====

image::foo.png[alt="Alt"]

Text below image 2
```

i.e. "Text below image 1" becomes "verbatim" text in asciidoc.

This has the strange side effect of creating a "<literallayout> in docbook, which is converted to a "/fvlisting" (fancyverb) in latex. Xelatex processes this using a temporary file, and chokes reading it back (/VerbatimInput) when it contains UTF-8 multibyte sequences starting with e.g. <0xC4> or <0xC5> (<0xC3> is fine).

This is an issue e.g. for Polish, see also #635

ciampix, 2019-03-12:
Which po4a version? Here: po4a --version
po4a versione 0.52. (Ubuntu 18.04)
and I do not see any mangled text... see below my po4a-normalize.output ....

```
|====
|First col
|Second col

image::foo.png[alt="Alt"]

Text below image 1
|====

image::foo.png[alt="Alt"]

Text below image 2
```

StefanBruens, 2019-03-12:
po4a 0.55
Probably:
https://github.com/mquinson/po4a/commit/5163d69ff591f4864935bdf5283cab782ec35583#diff-2d133d7e7eaed45848122efbb90e40e1R603
 "... except when we are in a table"

ciampix, 2019-03-12:
mmm time for update kicad-doc readme for minimum version of po4a (and also ubuntu)? Support for Ubuntu 14.04 is due to end in a few days. Given that 14.04 is missing of other libraries necessary for compilation of KiCad why not discard 14.04 and support only 16.04 altogether?
Let me know the working po4a minimum version anyway...

alfrede, 2019-03-15:
hi together, could it be that the failure come with version 0.54?

As i see in the release notes: https://github.com/mquinson/po4a/releases/tag/v0.54 there is a text which sounds: 

Remove the option for parsing AsciiDoc. There has been a standalone
module for nearly five years, released with v0.45. The AsciiDoc
option for the Text module has been deprecated since then.

i think this could be our Problem here in this case.

StefanBruens, 2019-08-08:
Why has this been closed - the issue is not fixed, the next time someone generates some translations the problem will pop up again.

nickoe, 2019-08-08:
Misclick I guess.