KiCad/kicad-doc #699: PCB_Calculator missing Board-Classes documentation

heitorPB, 2019-03-13:
PCB_Calculator docs (in all languages) have no description of the `Board-Classes` tab. 

Also, there's a very weird sentence in the `Electrical-Spacing` description: `The minimal Values which orientatd at the IPC-2221 too.` I'm not sure what this was supposed to mean.

alfrede, 2019-03-15:
Yes, i know it´s because i don't found any good Informations about the classes, the things i found where: 

https://www.eurocircuits.com/wp-content/uploads/EC-classification-ENGLISH-9-2018-V2.pdf

and:
https://www.protoexpress.com/blog/ipc-6012-ipc-600-standard-use/

but i don't really understand it and so i thought let it empty and somebody who understand it can write it. 

To the point with my sentence about the minimal values, okay it's not the best sentence but we (you) can change it or correct it. I am open for new content. 

The standard which i refer to is this: http://www.ipc.org/TOC/IPC-2221.pdf 

So please when anybody understand and find the right things in this pdf please correct the 2 points which are here discussed.

heitorPB, 2019-07-17:
What I found so far:

[IPC-6012B](http://www.ipc.org/TOC/IPC-6012B.pdf) says:
- there are 3 PCB Performance Classes
- there are 6 PCB types

No idea if this helps to document this part...