KiCad/kicad-doc #701: Why is there no Chinese Help Manual in Kicad 5.1.0？

taotieren, 2019-03-20:
Why is there no Chinese Help Manual in Kicad 5.1.0？

ciampix, 2019-03-20:
I see it here... http://docs.kicad-pcb.org/5.1.0/zh/
Where are you looking at?

taotieren, 2019-03-20:
I mean there is no Chinese manual in the packaged installation package (C:\Program Files\KiCad\share\doc\kicad\help),No zh folder

nickoe, 2019-03-20:
I assume you are only talking about the windows installer. This section needs to be updated:

https://github.com/KiCad/kicad-winbuilder/blob/d0546e9f0fa7925d4a81b0df91430bd6b559f04b/nsis/install.nsi#L385

taotieren, 2019-03-21:
@nickoe Thanks

taotieren, 2019-03-21:
What about the help manual on kicad under Linux and MacOS?

nickoe, 2019-03-21:
You need to check if they use implicit or explicit packaging rules. I saw that the debian package did package it. I am not sure about macos or the ubuntu ppa.

taotieren, 2019-03-22:
@nickoe OK, I will check if they are packaged, is there a packaged warehouse link? Packing under Windows I have added a Chinese installation profile and so on.

nickoe, 2019-03-22:
@taotieren Thank you. I added a comment on a minor issue. If you address it I can merge it.

nickoe, 2019-03-22:
@taotieren I think this is the source for the PPA packges, https://git.launchpad.net/~js-reynaud/+git/kicad-doc-pkg/tree/debian?id=5.1

@jsreynaud It does not look like zh is included.

jsreynaud, 2019-03-22:
Let me add it...

taotieren, 2019-03-22:
@nickoe  https://github.com/KiCad/kicad-winbuilder/pull/78

taotieren, 2019-03-22:
@jsreynaud @nickoe  @ciampix Thanks！

closed 2019-03-22 by nickoe