KiCad/kicad-doc #719: Can we have a 5.x branch?

craftyjon, 2019-04-20:
https://github.com/KiCad/kicad-doc/pull/709 is targeting 6.0 but it's pulled in to the 5.1.1 docs on the website.

nickoe, 2019-04-20:
That is on purpose. It seemed to me like the text does make it clear that it is a new feature in 6.0.

I don't think we gain much by maintaining multiple branches of the docs. There is so much work that will be removed from the master branch, and that already needs some care.

I propose to only have the master branch and write the text to tell the story about when a feature became available if needed.

nickoe, 2019-04-21:
@craftyjon Any more input on this?

craftyjon, 2019-04-21:
That is fine with me, I will keep it in mind when writing docs for other 6.0 features -- if it is clear that the policy is to have no branches for the docs, maybe in general we should avoid merging PRs that add documentation for new features until we are closer to the 6.0 release cycle

nickoe, 2019-04-22:
Well, I think we should just start documenting the 6.0 features now, and if we so decide that the docs are not compatible enough with 5.1 we could branch before making the next 5.1 tag OR just reuse it. 

As mentioned I am not strictly against creating a branch, I am just opposed if there is no one who is going to maintain it.

craftyjon, 2019-04-22:
OK, that makes sense to me.

closed 2019-04-22 by craftyjon

ciampix, 2019-05-20:
I can maintain it but just strictly in the mechanical part i.e. to assure that it compiles correctly. No control on contents at all. If you (all) think it is worth, I can do it.

nickoe, 2019-05-20:
@ciampix I still think time is better spent attempting to get the open pull requests closed, that be either rejecting it and/or try to apply it to t master. Then comes handling the current issue list.

sethhillbrand, 2019-06-21:
I would greatly appreciate a branch for major revisions that matches other sections of the project.  I understand @nickoe 's point about time management and it sounds like there are not enough hands to manage the current requests.  What do we need to do to train up more people on handling this repository?

nickoe, 2019-06-21:
@sethhillbrand I have asked various contributors earlier (probably only when we migrated to git) if they wanted write access to this repo. Most of them declined -- I think.

To move forward, I think we should really just focus on getting the master branch updated. There are quite a few issues open mentioning small features that needs to be documented.

If anyone feels like they are not heard, I suggest them to speak up.

sethhillbrand, 2019-06-21:
@nickoe that makes sense.  Thanks for the background.

It sounded in your previous comment like the issue was closing open pull requests first, which feels like a manpower problem.  If that's not the case, then please disregard my question about training additional people.

sethhillbrand, 2019-06-23:
@nickoe I should have asked explicitly in the previous comment, but what else is preventing the 5.1 branch for the documentation?  Without this, we are ending up with confused users like

https://forum.kicad.info/t/eeschema-group-bus-on-mac/17704

nickoe, 2019-06-23:
@sethhillbrand technically nothing. If it impedes documenting feature from master it may be better to do. I just won't expect any one updating it.

If people want the docs for a stable version they should just use the old docs.

The bus documentation may have slipped through to a 5.1.x tag, but it do say that this is available in v6.

> There are two types of bus in KiCad 6.0 and later: vector buses and group buses.

Although it may not be overly clear what part of the bus features are the old style and the new one.

But I am still hesitant to maintain a separate 5.1 branch as it causes way more work for the translators as well. If you think a 5.1 branch will help we can create it, but I am sure it will become quite stale anyway.

So my call is; please do document features on kicad master on the kicad-doc master. If we are to respin the docs for a 5.1 branch, we can still reuse the 5.1.x tag or branch at that point if we want to make the bus stuff clearer. but I have not seen a pull request for the later, so I have not made the branch.

nickoe, 2019-06-23:
@sethhillbrand So in other words; if your intention is to update the doc  for 5.1.3 to remove the new bus doc for that, please base it on tag 5.1.2 and I will very likely make a 5.1 branch.

sethhillbrand, 2019-07-15:
@nickoe I ran out of time to accomplish this effectively for 5.1.3, but I'm happy to take responsibility for submitting the PRs going forward.