KiCad/kicad-doc #722: Allow for multiple PCB layouts from single schematic in same project.

slimcdk, 2019-05-29:
The title pretty much covers what I'm requesting. Further explanation is that I want to create a generic part, that can be fitted onto different PCB layouts. 

For instance, I'm creating a mains voltage light controller. The circuit consists of exactly same components (symbols and footprints). However, the issue is that I want to be able create a circle and square PCB layout based on that same schematic. Changes to the schematic should affect both PCB layouts. The strucutre to do so is not important, but would be nice to see the implementation of tags.

Structures like this is what I have in mind:

Using tags:
./project_dir/
+-- main.pro
+-- main.sch
+-- main:layout1.kicad_pcb
+-- main:layout2.kicad_pcb

Using directories:
./project_dir
+-- main.pro
+-- main.sch
+-- pcbs
|  +-- layout1.kicad_pcb
|  +-- layout2.kicad_pcb

I have messed around with symbolic links to "duplicate" the project file and schematic. Or for changing 'active' PCB file, but neither solved the issue.

My project is here, if that clarifies anything:
https://github.com/slimcdk/libre-home-auto-components

slimcdk, 2019-05-29:
Addition:
I'm aware of the hierarchical sheet option in Kicad 5, but that is not what I'm seeking for.

CaninoDev, 2019-08-16:
This is more of a feature request for KiCAD itself and not an issue with KiCAD's documentation. I suggest making this request on [KiCAD's official repository](https://launchpad.net/kicad)

closed 2019-08-20 by nickoe