KiCad/kicad-doc #724: Documentation/guide about Library Management?

heitorPB, 2019-06-13:
My personal experience is that lib management can get very messy very fast. 

I found [this HackADay](https://hackaday.com/2017/05/18/kicad-best-practises-library-management/) article about it some years ago and I developed a similar workflow. Then I found [another article](https://forum.kicad.info/t/library-management-in-kicad-version-5/14636) in the forums about it.

The second link is really complete and also points to a script to help archive a project, which makes life easier. Many thanks to @poeschlr for the guide!

Is it possible to have an official guide/documentation about it? Or maybe move the forum post  to the official docs?

poeschlr, 2019-06-14:
The worflow used in the hackaday article does not work in v5 (one can not simply copy the cache lib to a different name and expect it to work. This would require updating the schematic references as well.)

heitorPB, 2019-06-14:
Indeed. I followed it before v5 came out and then I found your article :)