KiCad/kicad-doc #736: Translation - PCB_calculator - lost formatting

heitorPB, 2019-07-17:
Translations of PCB_calculator, section _2.4 TransLine_ are a bit screwed-up.

The english version has a nicely formatted list:
[image]

The translations lost their formatting:
[image]

heitorPB, 2019-07-20:
@ciampix your commit also fixes the list for the other languages, or just Italian?

I noticed the formatting is lost in all translations.

heitorPB, 2019-07-21:
Italian translation is still weird compared to the english one, but now is translated:

[image]

Other translations have the same issue. I suggest to reopen this issue.

ciampix, 2019-07-21:
I do not see any issue locally... how are you building it? Try update po4a, there were some improvement in the asciidoc parser...

ciampix, 2019-07-21:
Sorry for the delay ... I'm on vacation now and connecting via mobile with a very weak & slow connection...

heitorPB, 2019-07-21:
I'm building like this:

```
mkdir build && cd build
cmake .. # and I got rid of some languages that I don't have the fonts
make -j 5
```

But those screenshots were taken from the [online docs](http://docs.kicad-pcb.org/master/it/pcb_calculator/pcb_calculator.html#transline).

ciampix, 2019-07-21:
Ok and you see the same results locally or not?

heitorPB, 2019-07-21:
Weird, locally is also fine. I checked the other translations and are also fine in my computer (clean build). The problem is in the online docs.

Do you know how is compiled to put it online?

ciampix, 2019-07-22:
Ok that's a (relatively smaller) problem of the web devs like perhaps @nickoe ...

nickoe, 2019-07-28:
The website builder uses po4a version 0.47.

nickoe, 2019-07-29:
@ciampix and @heitorPB what version are you using?

heitorPB, 2019-07-29:
I'm using po4a version 0.56

ciampix, 2019-07-30:
I'm using po4a version 0.52 ...

heitorPB, 2019-08-07:
I built the docker image in `utils/docker/Dockerfile.kicad-doc-builder-base` and I got po4a version 0.55. @nickoe how old is your docker image? Or is there a reason to use po4a version 0.47?

commit d37114f2 (2019-07-20): Update Italian translation of pcb_calculator, fixes #736