KiCad/kicad-doc #739: Tag 5.1.3

nickoe, 2019-07-24:
I will tag this repo sunday (European time zone) unless someone has some changes they want included for 5.1.3, please mention it and we can extend it a bit. I see a lot of contributions, especially translations has been ticking in lately. Thank you.

nickoe, 2019-07-24:
@sethhillbrand if you still habe an intention and time to make the pr you mentioned that may branch 5.1.3 from master, please do so befpre sunday.

sethhillbrand, 2019-07-24:
Thanks @nickoe but I feel that would split the translations from the english docs, which I'd prefer not to do.  I will submit the PR after 5.1.3 is tagged.

heitorPB, 2019-07-28:
Sorry for not seeing this issue before. I opened #736 and there is a problem with the translations of PCB_Calculator documentation. The online master version is a scrambled and we can't reproduce locally.

nickoe, 2019-08-09:
Closing this as it was tagged and I just tagged 5.1.4 as well.

closed 2019-08-09 by nickoe