KiCad/kicad-doc #741: Build of documentation for language zh is failing on tag 5.1.3

tijuca, 2019-07-29:
I'm unable to build all documentation modules for Chinese on tag 5.1.3.
```

$ LANG= cmake .. -DLANGUAGES="zh"
-- Found DBLATEX: /usr/bin/dblatex (found version "0.3.10") 
-- Found PO4A: /usr/bin/po4a (found version "0.55") 
-- Found ASCIIDOC: /usr/bin/asciidoc (found version "8.6.10") 
-- Found A2X: /usr/bin/a2x (found version "8.6.10") 
-- Found Git: /usr/bin/git (found version "2.20.1") 
-- On branch HEAD with tag 5.1.2-15-gd63945e4
-- Setting package version to 5.1.2-15-gd63945e4
-- Configuring done
-- Generating done
-- Build files have been written to: /home/carsten/gitprojects/kicad-upstream/kicad-doc/build
```
And building the stuff ...
```
$ LANG= make -j4
Scanning dependencies of target cvpcb_translate_zh
Scanning dependencies of target getting_started_in_kicad_translate_zh
Scanning dependencies of target gerbview_translate_zh
Scanning dependencies of target eeschema_translate_zh
[  2%] Generating zh/gerbview.adoc
[  2%] Generating zh/cvpcb.adoc
[  2%] Generating zh/getting_started_in_kicad.adoc
[  3%] Generating zh/eeschema.adoc
[  3%] Built target gerbview_translate_zh
Scanning dependencies of target idf_exporter_translate_zh
Your input po file /home/carsten/gitprojects/kicad-upstream/kicad-doc/src/cvpcb/po/zh.po seems outdated (The amount of entries differ between files: 230 is not 135
). Please consider running po4a-updatepo to refresh it.
[  4%] Generating zh/idf_exporter.adoc
[  5%] Generating zh/cvpcb_footprint_libraries_tables.adoc
...
[ 93%] Built target pl_editor
[ 94%] Built target gui_translation_howto_epub_zh
Scanning dependencies of target gui_translation_howto
[ 94%] Built target gui_translation_howto
[ 95%] Built target pcbnew_pdf_zh
Scanning dependencies of target pcbnew_epub_zh
[ 96%] Generating zh/pcbnew.epub
a2x: ERROR: "dblatex" -t pdf -p "/etc/asciidoc/dblatex/asciidoc-dblatex.xsl" -s "/etc/asciidoc/dblatex/asciidoc-dblatex.sty" -P latex.output.revhistory=0 -P doc.publisher.show=0 -s /home/carsten/gitprojects/kicad-upstream/kicad-doc/CMakeSupport/pdf-cover-dblatex.sty -b xetex -p /home/carsten/gitprojects/kicad-upstream/kicad-doc/src/eeschema/../../xsl/dblatex-pdf-zh.xsl -o /home/carsten/gitprojects/kicad-upstream/kicad-doc/build/src/eeschema/zh/eeschema.pdf  "/home/carsten/gitprojects/kicad-upstream/kicad-doc/build/src/eeschema/zh/eeschema.xml" returned non-zero exit status 1
make[2]: *** [src/eeschema/CMakeFiles/eeschema_pdf_zh.dir/build.make:61: src/eeschema/zh/eeschema.pdf] Error 1
make[1]: *** [CMakeFiles/Makefile2:920: src/eeschema/CMakeFiles/eeschema_pdf_zh.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 97%] Built target pcbnew_epub_zh
make: *** [Makefile:140: all] Error 2
```
O.k. what's wrong in detail:
```
$ dblatex -d -t pdf -p "/etc/asciidoc/dblatex/asciidoc-dblatex.xsl" -s "/etc/asciidoc/dblatex/asciidoc-dblatex.sty" -P latex.output.revhistory=0 -P doc.publisher.show=0 -s /home/carsten/gitprojects/kicad-upstream/kicad-doc/CMakeSupport/pdf-cover-dblatex.sty -b xetex -p /home/carsten/gitprojects/kicad-upstream/kicad-doc/src/eeschema/../../xsl/dblatex-pdf-zh.xsl -o /home/carsten/gitprojects/kicad-upstream/kicad-doc/build/src/eeschema/zh/eeschema.pdf  "/home/carsten/gitprojects/kicad-upstream/kicad-doc/build/src/eeschema/zh/eeschema.xml"
Build the book set list...
Build the listings...
XSLT stylesheets DocBook - LaTeX 2e (0.3.10)
===================================================
No localization exists for "zh" or "". Using default "en".
No localization exists for "zh" or "". Using default "en".
No localization exists for "zh" or "". Using default "en".
No localization exists for "zh" or "". Using default "en".
No localization exists for "zh" or "". Using default "en".
No localization exists for "zh" or "". Using default "en".
No localization exists for "zh" or "". Using default "en".
No localization exists for "zh" or "". Using default "en".
No localization exists for "zh" or "". Using default "en".
No localization exists for "zh" or "". Using default "en".
No localization exists for "zh" or "". Using default "en".
No localization exists for "zh" or "". Using default "en".
No localization exists for "zh" or "". Using default "en".
No localization exists for "zh" or "". Using default "en".
No localization exists for "zh" or "". Using default "en".
No localization exists for "zh" or "". Using default "en".
[... 111 lines cut ...]
eeschema.tex:7836: leading text: }
eeschema.tex:7836: Undefined control sequence.
eeschema.tex:7836: leading text: }
eeschema.tex:7836: Undefined control sequence.
eeschema.tex:7836: leading text: }
eeschema.tex:7836: Undefined control sequence.
eeschema.tex:7836: leading text: }
eeschema.tex:7836: Undefined control sequence.
eeschema.tex:7836: leading text: }
eeschema.tex:7836: Undefined control sequence.
eeschema.tex:7836: leading text: }
eeschema.tex:7836: Undefined control sequence.
eeschema.tex:7836: leading text: }
eeschema.tex:7836: Undefined control sequence.
eeschema.tex:7836: leading text: }
eeschema.tex:7836: Undefined control sequence.
eeschema.tex:7836: leading text: }
/tmp/tmp5Ma9MT not removed

A possible reason for transformation failure is invalid DocBook
(as reported by xmllint)

Error: xelatex compilation failed
```
I guess this is the same problem as happen earlier within other releases similary. Some Chinese characters (previously it was Japanese) confusing asciidoc. And yes, if I'm going back to to ID 7c863927 the build is succeeding again.

nickoe, 2019-07-29:
I see the same when I tried to update to debian testing, when I tried to update to get a newer po4a. Reverting 642cc5e8576ac3eb788814b7f82284e462d0a204 fixes it, so it is very likely something introduced in the `src/eeschema/po/zh.po`, although it did build on the older environment as seen by the docker build linked on the pull request.

nickoe, 2019-07-29:
@taotieren @ciampix

tijuca, 2019-07-29:
No, the problem is not po4a or the updated l10n content for Eeschema in Chinese, the problem is still then asciidoc that is doing the preparation for Xelatex not good enough. I've sit last year with the Debian Maintainer for latex together for over two hours to debug the issue that was happen for Japanese and we stopped in the end on asciidoc. Some latex packages are more picky about the provided source now, especially for Chines and Japanese.

So it is how it is now, that's show again that more automatic testing is needed, running some Ubuntu LTS is not enough. Unfortunately I've spent until now several hours on that issue that preventing me to do an upload to Debian. I'd like to just modify the "problematic" parts in commit 7c86392 so most of the translations are usable.

nickoe, 2019-07-29:
@tijuca it was tested on debian stretch and ubuntu trusty.

tijuca, 2019-07-29:
I know, Ubuntu Trusty is even older than Stretch. ;)
The problematic things are happen so far I see on recent versions of latex.

It's obvious that the problem is a character '}' that is there for some unknown reason in the generated tex file.

```
eeschema.tex:7836: Undefined control sequence.
eeschema.tex:7836: leading text: }
```

```
7823 \begin{center}
7824 
7825 \noindent\begin{minipage}[c]{\linewidth}
7826 \begin{center}
7827 \imgexists{images/opamp_symbol.png}{{\imgevalsize{images/opamp_symbol.png}{\includegraphics[width=\imgwidth,height=\imgheight,keepaspectratio=true]{images/     opamp_symbol.png}}}}{Generic operational amplifier symbol}\end{center}
7828 \end{minipage}
7829 
7830 
7831 \end{center}
7832  \begin{fvlisting}
7833 \VerbatimInput[fontfamily=tt,]{tmplst-idm5010}
7834 \end{fvlisting}
7835 %
7836 }\tabularnewline
     ^  here
7837 \hline
7838 \end{longtable}\endgroup%
7839 \restoretablecounter%
7840 }
```
And this mus be related to line [5357](https://github.com/KiCad/kicad-doc/blob/master/src/eeschema/po/zh.po#L5357). If I remove that single line the build is succeeding. I'll probably will remove the whole section within that po file so I can kick out the Debian package tomorrow after some more testing.

nickoe, 2019-07-29:
If I remove the double space in the translation it seems to build as well.

tijuca, 2019-07-29:
Can you providing a diff?

tijuca, 2019-08-01:
"Solved" by adding the workaround to the tree.

closed 2019-08-01 by tijuca

tijuca, 2019-08-03:
Why was this small fix not added to the main branch? Why create a new branch for this? My local build was breaking again after pulling from GitHub and I was wondering why. :disappointed: 

It would've been better to ping the original translator to review than to create "one more branch" or bring this to discussion on the mailing list.

@taotieren 
Could you please give a comment on the issue!?

As we will go another round for a new tag 5.1.4 there will be time to fix this finally.

taotieren, 2019-08-03:
@tijuca Hi,I translated and tested it in the local environment (Manjaro Linux), and there was no compilation error. The translated document was also displayed normally. My locally compiled document (https://gitee.com/KiCAD-CN/KiCad- Doc_build).

taotieren, 2019-08-03:
I want to know under what circumstances latex will have compilation errors, we can list some known issues to form a translation documentation. This will prevent these errors from being introduced later in the translation.

tijuca, 2019-08-03:
I was expecting that you have tested your work. :+1: 
I mean more if the workaround for this particular string is also working / readable within Chinese.

I usually have running a Debian testing environment and do all of my daily work here. From time tome I build the tree besides my packaging work for Debian. So I will cover some building issue also from time to time.

The problem here, as before for Japanese, isn't latex. The problem is asciidoc which is doing a bad preparation now for latex. The recent latex toolchain did get a lot of fixups and extensions in the past two years, but this also is now more critical about the correctness of the input data to work properly for Japanese and Chinese. The only solution to detect building issues is to use a recent toolchain too.

nickoe, 2019-08-03:
All pull requests are tested to some extend. It works on some platforms and some not.

My test shows that the old docker image based on some older debain works and when I updated that base builder image to the latest tag where it started to fail.

taotieren, 2019-08-03:
I will test it later in the local build environment and the VPS (Ubuntu 1904) build environment.

taotieren, 2019-08-03:
@nickoe  @tijuca The Simplified Chinese translation document has been updated. Please do multi-compilation environment test compile can pass。

tijuca, 2019-08-03:
I've build current master HEAD at 06cfd3067f3195f19ef43652f4228ca920a1dbc5 successful within Debian testing. Looks good! Thanks!

ciampix, 2019-08-04:
> The problem here, as before for Japanese, isn't latex. The problem is asciidoc which is doing a bad preparation now for latex. The recent latex toolchain did get a lot of fixups and extensions in [...]

Did anyone check or file a bug report on the asciidoc project page for this?

tijuca, 2019-08-04:
In my eyes this is more or less useless as asciidoc is Python2 which will get obsolete in the near future given the support for Python2 isn't alive for longer time. So in midterm we need to find a solution at all.
```
$ file /usr/bin/asciidoc /usr/bin/a2x
/usr/bin/asciidoc: a /usr/bin/python2 script, ASCII text executable
/usr/bin/a2x:      a /usr/bin/python2 script, ASCII text executable
```

The options i see is to use asciidoctor(-pdf) or reStructured Text and other toolchains. Markdown isn't a real option.

heitorPB, 2019-08-04:
Another problem with translations is the one I reported in #736, for all languages. Locally it builds fine, on the website, it's different.

#638 says something to migrate from asciidoc to asciidoctor.

tijuca, 2019-08-05:
#736 is a complete different problem, please don't mix up these things between the issues! I suggest your clear up the used versions and compare which combination(s) are not working within your original report.

ciampix, 2019-08-07:
> In my eyes this is more or less useless as asciidoc is Python2 which will get obsolete in the near future given the support for Python2 isn't alive for longer time. So in midterm we need to find a [...]

Actually there is a python 3 version of asciidoc around that the devs are testing and updating actively... one of these days it will become the mainstream asciidoc python version but I do not know what are they waiting for...

StefanBruens, 2019-08-07:
This *is* a bug in po4a, as it inserts the two spaces in front, which is asciidoc markup for "verbatim", see #697.

commit a4daa042 (2019-07-29): zh po hack

Fixes #741
commit 691d595f (2019-08-03): zh po hack

Uses the fix propsed at
https://github.com/KiCad/kicad-doc/issues/741#issuecomment-516168156

Fixes #741