KiCad/kicad-doc #762: Wrong German Menue Translation for pcbnew

Jan-vi, 2019-10-07:
Sorry if this is wrong here - if so pls redirect to the Kicad code dev group
Pcbnew shows a menu item with Place->Dimension.  For German translation, there are two diffrent expressions in use for "dimension".  The German translation "Einfügen->Abmessungen"  should change in "Einfügen->Bemaßung". For Austria and Switzerland "Bemassung" is ok (becouse of missing "scharfes Ess" in their character set)  and will be also accepted for Germans. "Abmessungen" is also ok to understand after using once, but correct menue items makes the Kicad much more intuitive for beginners who consider to change from other EDA. "Abmessung" may be accepted for e.g. length=30mm if used for associative drawing input while "Bemaßung" describes the use of arrows graphics in all styles. For examples see the lesson here: 

https://www.hs-anhalt.de/fileadmin/Dateien/FB6/personen/voigt_st/Lehrunterlagen/05_CAD/04_techZeich_Masseintragungen.pdf 

Even Autocad is translating well:
https://knowledge.autodesk.com/de/support/autocad/getting-started/caas/CloudHelp/cloudhelp/2019/DEU/AutoCAD-Core/files/GUID-9810D35D-E975-49E2-88EA-51FE4ACC92DB-htm.html

Application: Pcbnew
Version: (5.1.4-87-g78af1daa2)-1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.61.1 OpenSSL/1.1.1 (WinSSL) zlib/1.2.11 brotli/1.0.6 libidn2/2.0.5 libpsl/0.21.0 (+libidn2/2.1.1) nghttp2/1.34.0
Platform: Windows 8 (build 9200), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8)
    Boost: 1.68.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.61.1
    Compiler: GCC 8.2.0 with C++ ABI 1013

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=OFF
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

poeschlr, 2019-10-07:
translations are handled in https://github.com/KiCad/kicad-i18n

closed 2019-10-07 by poeschlr

Jan-vi, 2019-10-07:
thx Rene, I am going to post there

poeschlr, 2019-10-07:
I am not 100% certain issues are handled on github at all. It could be that translators have a separate mailing list. Maybe check the contribution guidelines on the i18n repo or on the kicad website

poeschlr, 2019-10-07:
All i could find is https://forum.kicad.info/t/rc2-call-for-translators/10839 and https://forum.kicad.info/t/how-can-i-help-improve-kicad/15217 so it seems like github might indeed be the best option.