KiCad/kicad-doc #763: PCB Calculator Units of Measure

LarryJoy, 2019-10-18:
**Regulators tab**
For R1 change "KOhm" to "kΩ".
For R2 change "KOhm" to "kΩ".
For iadj change "uA" to "µA".
Note: When spelled out (in English) units of measure are considered common nouns and are NOT capitalized.
**Track Width tab**
For Temperature rise change "deg C" (degree coulomb?) to "°C".
For Resistivity change "Ohm-meter" to "Ω•m" [Note: hyphens are not used with SI letter symbols.]
For External layer traces:
1) Change Resistance "Ohm" to "ohm" or better/best to "Ω".
2) Change Voltage drop "Volt" to "volt" or better/best to "V".
3) Change Power loss "Watt" to "watt" or better/best to "W".
For Internal layer traces:
1) Change Resistance "Ohm" to "ohm" or better/best to "Ω".
2) Change Voltage drop "Volt" to "volt" or better/best to "V".
3) Change Power loss "Watt" to "watt" or better/best to "W".
2nd paragraph
1) Change "35A" to "35 A". [Note: There is always a space between the value (number) and the following SI letter symbol.
2) Change "17.5A" to "17.5 A".
3) Change "100 deg C" to "100 °C".
4) Change "400mil" to "400 mil".
5) Change "10mm" to "10 mm".
3rd paragraph
1) In dT change "deg C" to "°C".
**Electrical Spacing tab**
1) Where it is stated "Voltage > 500V" change "500V" to "500 V".
2) For the column with voltage ranges change "*V" to "* V".
**TransLine tab**
For Electrical Parameters:
1) Z0 "Ohm" should be "Ω" and "KOhm" should be "kΩ".
2) Ang_l "Radian" should be "rad" and "Degree" should be "degree" or better/best "°".
**RF Attenuators tab**
1) Zin "Ohms" should be "ohms" or better/best "Ω".
2) Zout "Ohms" should be "ohms" or better/best "Ω".
Values:
1) R1 "Ohms" should be "ohms" or better/best "Ω".
2) R2 "Ohms" should be "ohms" or better/best "Ω".
3) R3 "Ohms" should be "ohms" or better/best "Ω".
For "Attenuation is 6dB". "6dB" should be "6 dB".
**Board Classes tab**
In the upper left corner dimensions box "um" should be "µm".
--Regards, Larry

nickoe, 2019-10-18:
Are you going to submit a pull-request for this?

LarryJoy, 2019-10-18:
I was just following the instructions on the KiCad EDA documents page where it states to submit feedback/comments go to https://github.com/kicad/kicad-doc/issues. I signed in with my identification (name and password) and entered my feedback/comments. In the past I have had problems with the pull-request process because of the nomenclature used. If I should submit a pull-request please explain how to do it or refer me to instructions.--Larry

| 
| 
|  | 
KiCad EDA

A Cross Platform and Open Source Electronics Design Automation Suite
 |

 |

 |

marekr, 2019-10-21:
Unfortunately this is feedback for the application which requries using launchpad.

poeschlr, 2019-10-21:
To make it easier here the link to the launchpad bug tracker https://bugs.launchpad.net/kicad

nickoe, 2019-11-15:
@LarryJoy Thank you for your report and interest in KiCad, but as the others have mentioned. The details you state above are for the application itself and not the documentation. Please open a bug for it on launchpad.net if you think it is still relevant.

closed 2019-11-15 by nickoe