KiCad/kicad-doc #768: Ubuntu - ppa user or team does not exist

kdewilde, 2019-11-26:
Hi

I don't know if this is the correct location to log this, but when I got to the [download page](http://kicad-pcb.org/download/ubuntu/) and execute the cli command

`sudo add-apt-repository --yes ppa:js-reynaud/kicad-5.1`

I get the following error

`Cannot add PPA: 'ppa:~js-reynaud/ubuntu/kicad-5'.
ERROR: '~js-reynaud' user or team does not exist.`

nickoe, 2019-11-26:
This is not the right place, this should be on launchpad.  But did you add that tilde yourself?

closed 2019-11-26 by nickoe

kdewilde, 2019-11-26:
I did not add that myself. I copy/pasted the command from the download page.

What do you mean with launchpad? could you provide me with a link to that?

sethhillbrand, 2019-11-26:
Here is the PPA link https://launchpad.net/~js-reynaud/+archive/ubuntu/kicad-5.1

If you have issues, log a bug report at https://bugs.launchpad.net/kicad.