KiCad/kicad-doc #769: GitLab Documentation Migration

sethhillbrand, 2019-11-27:
Hi Documenters-

As part of the Great KiCad GitLab migration of 2019, I am testing the movement of open issues and PRs from GitHub to GitLab.

Here's what we know so far:

- Open Pull Requests will transfer to GitLab without moving the source of the PR.  That means pull requests will become GitLab Merge Requests that are referenced to the user's GitHub repository
- Open Issues will transfer to GitLab issues
- Closed/Resolved PRs and Issues will also transfer

In order to facilitate the transfer, it will be helpful if you can do two things:
- Open a GitLab account (these can use your GitHub OAuth)
- **Set your GitHub email to public**
- Let me know who the responsible parties are for this repository

The second point is critical because the migration script will only link users whose e-mail addresses match.  If the script can't see your GitHub e-mail address, the issue will be tied to the KiCad-Bot that performs the migration.

The timeline for the actual migration will be determined by the responsible parties from bullet 3 above but it would greatly assist my testing for bugs if you could do the first two steps above as soon as possible.

Thank you-
Seth

eabase, 2019-11-27:
Why are you moving to GitLab? 

IMO I  think this place has been working just fine. But now you want people to start signing up for GitLab instead.

sethhillbrand, 2019-11-27:
Many issues touch source, documentation and packaging. Having all elements in a single location makes managing/fixing/updating easier for maintainers and promotes greater collaboration

heitorPB, 2019-11-28:
The GitLab repo is here> https://gitlab.com/kicad/documentation/kicad-doc/

In the [Contributors page](https://gitlab.com/kicad/documentation/kicad-doc/-/graphs/master) i can see people (myself included) that used the noreply mail from GitHub. I guess they will not have their GitLab account automatically linked? Is there a way to change this?

sethhillbrand, 2019-11-28:
Update your e-mail setting under preferences:
[image]

Scroll down and make sure your e-mail address is not private:
[image]

In another week or so, I'll update the GitLab mirror and check to see how many are still missing.

nickoe, 2019-11-28:
@sethhillbrand me and ciampix

ciampix, 2019-11-28:
Thanks @nickoe! I was already member of gitlab. I use the same email that I use in github. I set it public (hope I can unset that option) I also enabled the 2FA (but think that it is an overkill) but still I am not a member of kicad-doc and kicad-web ...

chschlue, 2019-11-28:
Semi-OT:
Why do you consider 2FA overkill?
KiCAD wouldn't be be the first project troubled by compromised accounts.
(Ubuntu and Gentoo come to mind)

ciampix, 2019-11-29:
Because for the documentation and translations I do not see a lot of potential security problems. But that was an unuseful comment since I, as I said, I already enabled 2FA...
PS: I forgot the kicad-i18n of course, I would like to be part of that repo devs group also, as I was in github. I am now part of the kicad-doc (thanks!).

poeschlr, 2019-11-29:
Well 2FA on gitlab requires a device that is capable of running the required software. Which is either a dedicated hardware stick (costs 50$) or an app on a smartphone (Requires a smartphone capable of running that software costs >50$). 
I am one example of somebody not in possession of either such a device meaning if the kicad project would require this for all groups then i would no longer be able to help out. (The library section already opted against making 2FA a requirement. See mailing list for details.)

I am sure there are many more people in a similar situation as i am so it really needs to be evaluated carefully where it makes sense and where it does not. 

Also remember 2FA only guarantees that the person making a contribution is the holder of an account (or more precisely the holder of all required devices and information required to identify as that account) It does not guarantee that this person is trustworthy. It does not even guarantee that the person is who they claim to be (that would require part of the 2FA registration to involve a direct person to person meeting or some trusted source that can verify the personal information.)

chschlue, 2019-12-01:
> @chschlue Well 2FA on gitlab requires a device that is capable of running the required software. Which is either a dedicated hardware stick (costs 50$) or an app on a smartphone (Requires a smartphone capable of running that software costs >50$).

They're available much cheaper these days.

> I am one example of somebody not in possession of either such a device meaning if the kicad project would require this for all groups then i would no longer be able to help out. (The library section already opted against making 2FA a requirement. See mailing list for details.)

I wasn't talking about making 2FA a requirement, but why discourage its use?

> Also remember 2FA only guarantees that the person making a contribution is the holder of an account (or more precisely the holder of all required devices and information required to identify as that [...]

True, but beyond the point.
Simply put, 2FA makes it more likely that the person committing tomorrow is still the same person that was given access yesterday.
It's certainly not a panacea but it doesn't hurt either.

sethhillbrand, 2019-12-01:
Documentation migration has been updated.

Please look over the resulting repository at https://gitlab.com/kicad/website/kicad-documentation

If there are no further changes requested, I would like to schedule the official migration date for this month.

nickoe, 2019-12-01:
I consider this migrated.

closed 2019-12-01 by nickoe